mavlink/qgroundcontrol #1458: Setup is refused even though only one system is connected

LorenzMeier, 2015-04-15:
This happens if only one system is connected. The issue is real, but I don't understand the mechanism behind it yet.

[image]

LorenzMeier, 2015-04-15:
@tumbili For your reference.

DonLakeFlyer, 2015-04-15:
It's even worse than that. Something recent is causing it to crash on OS X when you do this!

DonLakeFlyer, 2015-04-15:
Looks like that crash is just when you are testing with MockLink. I can't repro on OSX. Have to try Ubuntu. Is this with a USB connection?

tumbili, 2015-04-15:
@DonLakeFlyer Hi Don, yes this is with USB connection on Ubuntu 14.04. Even if I restart everything I always get the device files "ttyACM0" and "ttyACM1".

DonLakeFlyer, 2015-04-15:
Fixed

closed 2015-04-16 by DonLakeFlyer