mavlink/qgroundcontrol #1467: Need sensible default initial size for QGC Window

dogmaphobic, 2015-04-16:
When you bring QGC up for the first time, or whenever the settings are reset, it opens up with a humongous size. On Mac OS it fills an entire monitor, even if the monitor is a 30" display. On Windows it's even worse. If you have multiple monitors, it will span them all creating a truly humongous window.

I would suggest making its initial size based on the screen size. That is, if you are running 1024x768 monitor (do those still exist?), it makes sense to fill up the screen. On larger screens, limit to something more sensible.

thomasgubler, 2015-04-17:
I guess regarding window size it should default to the standard on each platform. Spanning across screens is of course wrong.

dagar, 2015-04-17:
I personally think starting maximized (single screen) would be a good default, but I only use Ubuntu and Windows. This is certainly the only way I'd want it in the field. It sounds like something else might make sense on OS X though.

dogmaphobic, 2015-04-17:
> I personally think starting maximized (single screen) would be a good default,

I guess that would be the case for most real use cases. Out in the field, with a laptop or something equivalent. It's not a good default for the desktop though, specially during a heavy development phase. Having it come up full screen on a large display, or multiple displays as it is the case in Windows, is not a good default. Maybe there should be a separate setting between release and development builds.

dagar, 2015-04-17:
What about default to fullscreen on one monitor, but remember the last resize?

dogmaphobic, 2015-04-17:
It already remembers the last size. This is primarily an issue when you reset the settings (and don't have a previous size to use). On subsequent runs, it will use whatever it was when you closed it.

DonLakeFlyer, 2015-04-18:
I'd say start at minimum supported resolution which is 1024x768. Then you can adjust from there.

dogmaphobic, 2015-04-22:
Closed as it was resolved by #1492

closed 2015-04-22 by dogmaphobic