mavlink/qgroundcontrol #1469: Unable to build on Ubuntu 14.04

tumbili, 2015-04-16:
When I try to build from source I get the following error:
/usr/bin/ld: build-release/obj/QGCMAVLinkInspector.o(.text+0x8000000a903): reloc against `.LC252': error 2
/usr/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
make[1]: *** [release/qgroundcontrol] Error 1
make: *** [sub-QGCApplication-pro-make_first-ordered] Error 2

tumbili, 2015-04-16:
I tried "make clean" and "git clean -dfx" but that didn't help

DonLakeFlyer, 2015-04-16:
I don't think this is caused by QGC. Likely something on your box. A linux person may need to help you out. My Ubuntu builds are still working fine from master. Once just went through TeamCIty a few minutes ago.

DonLakeFlyer, 2015-04-16:
What are you using for the qmake command line?

dagar, 2015-04-16:
Current master is still building for me on Ubuntu 14.04 64bit with the qt54 ppa and gcc 4.8

/opt/qt54/bin/qmake
make -j4

DonLakeFlyer, 2015-04-16:
You need to add -r to your qmake command line otherwise you may get incorrect makefiles.

tumbili, 2015-04-17:
@DonLakeFlyer Thanks, it works with the "-r" option.

closed 2015-04-17 by tumbili