mavlink/qgroundcontrol #1475: Waypoint planning for fixed wing - add descend degree and distance

severinleuenberger, 2015-04-18:
When planning missions, one of the bigger issues, is the missing climb/descend degree and distance between waypoints. I was flying with 2 military pilots for their bachelor thesis and they had a lot of difficulties by planning while missing those 2 informations. 

Maybe even the AZ (heading) would be great. But Distance and degree up or down are a must on the long term

Best regards, 
Severin 

[image]

severinleuenberger, 2015-05-11:
We just finished 80 flights on operating for autolanding with 2 swiss military pilots. 
It was very difficult do plan operations without having the distance and the degrees of climb/descend to plan our mission. Please do not forget this enhancement. Would be a very important one for mission planning.

severinleuenberger, 2015-08-14:
...if you have time to build on QGC, this enhancement would be ery helpflul for the users who plan missions.

DonLakeFlyer, 2015-12-03:
Distance and Azimuth between waypoints is now there. Still need to add climb.

DonLakeFlyer, 2015-12-04:
You now have distance, azimuth and altitude change from previous waypoint.

closed 2015-12-04 by DonLakeFlyer