mavlink/qgroundcontrol #1480: Airframe config link re-connect crashes application

LorenzMeier, 2015-04-19:
This is a fresh (completely clean) build off current master:

```
Process:         qgroundcontrol [32449]
Path:            /Users/USER/*/qgroundcontrol.app/Contents/MacOS/qgroundcontrol
Identifier:      org.qgroundcontrol.qgroundcontrol
Version:         2.0.0 (227)
Code Type:       X86-64 (Native)
Parent Process:  Qt Creator [30464]
Responsible:     Qt Creator [30464]
User ID:         501

Date/Time:       2015-04-19 11:44:29.269 +0200
OS Version:      Mac OS X 10.9.5 (13F1066)
Report Version:  11
Anonymous UUID:  0BE00B61-2D1F-543C-38F9-2BD7ACFA6ECA

Sleep/Wake UUID: D16301AB-98EE-45F8-9231-FB885B717103

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: EXC_I386_GPFLT

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qgroundcontrol.qgroundcontrol	0x00000001000fc426 LinkManager::createConnectedLink(LinkConfiguration*) + 22
1   org.qt-project.QtCore         	0x0000000101813114 QSingleShotTimer::timerEvent(QTimerEvent*) + 132
2   org.qt-project.QtCore         	0x0000000101805300 QObject::event(QEvent*) + 48
3   org.qt-project.QtWidgets      	0x0000000102bcedbb QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
4   org.qt-project.QtWidgets      	0x0000000102bd2110 QApplication::notify(QObject*, QEvent*) + 8192
5   org.qt-project.QtCore         	0x00000001017d9fb3 QCoreApplication::notifyInternal(QObject*, QEvent*) + 115
6   org.qt-project.QtCore         	0x000000010182e3f6 QTimerInfoList::activateTimers() + 1270
7   libqcocoa.dylib               	0x0000000104505d12 QCocoaEventDispatcherPrivate::activateTimersSourceCallback(void*) + 18
8   com.apple.CoreFoundation      	0x00007fff8493f5b1 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
9   com.apple.CoreFoundation      	0x00007fff84930c62 __CFRunLoopDoSources0 + 242
10  com.apple.CoreFoundation      	0x00007fff849303ef __CFRunLoopRun + 831
11  com.apple.CoreFoundation      	0x00007fff8492fe75 CFRunLoopRunSpecific + 309
12  com.apple.HIToolbox           	0x00007fff8dc90a0d RunCurrentEventLoopInMode + 226
13  com.apple.HIToolbox           	0x00007fff8dc907b7 ReceiveNextEventCommon + 479
14  com.apple.HIToolbox           	0x00007fff8dc905bc _BlockUntilNextEventMatchingListInModeWithFilter + 65
15  com.apple.AppKit              	0x00007fff871ea24e _DPSNextEvent + 1434
16  com.apple.AppKit              	0x00007fff871e989b -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 122
17  com.apple.AppKit              	0x00007fff871dd99c -[NSApplication run] + 553
18  libqcocoa.dylib               	0x000000010450692d QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 2189
19  org.qt-project.QtCore         	0x00000001017d75ad QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381
20  org.qt-project.QtCore         	0x00000001017da58a QCoreApplication::exec() + 346
21  org.qgroundcontrol.qgroundcontrol	0x00000001001404c9 main + 297
22  org.qgroundcontrol.qgroundcontrol	0x0000000100009d64 start + 52

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8ef3c662 kevent64 + 10
1   libdispatch.dylib             	0x00007fff88723421 _dispatch_mgr_invoke + 239
2   libdispatch.dylib             	0x00007fff88723136 _dispatch_mgr_thread + 52

Thread 2:: QProcessManager
0   libsystem_kernel.dylib        	0x00007fff8ef3b9aa __select + 10
1   org.qt-project.QtCore         	0x00000001017782af QProcessManager::run() + 143
2   org.qt-project.QtCore         	0x00000001015d0393 QThreadPrivate::start(void*) + 339
3   libsystem_pthread.dylib       	0x00007fff8df6e899 _pthread_body + 138
4   libsystem_pthread.dylib       	0x00007fff8df6e72a _pthread_start + 137
5   libsystem_pthread.dylib       	0x00007fff8df72fc9 thread_start + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8ef37a1a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff8ef36d18 mach_msg + 64
2   com.apple.CoreFoundation      	0x00007fff84930f15 __CFRunLoopServiceMachPort + 181
3   com.apple.CoreFoundation      	0x00007fff84930539 __CFRunLoopRun + 1161
4   com.apple.CoreFoundation      	0x00007fff8492fe75 CFRunLoopRunSpecific + 309
5   com.apple.AppKit              	0x00007fff8738a05e _NSEventThread + 144
6   libsystem_pthread.dylib       	0x00007fff8df6e899 _pthread_body + 138
[... 90 lines cut ...]
10  libsystem_pthread.dylib       	0x00007fff8df72fc9 thread_start + 13

Thread 13:: QThread
0   libsystem_kernel.dylib        	0x00007fff8ef3b9aa __select + 10
1   org.qt-project.QtCore         	0x000000010182b134 qt_safe_select(int, fd_set*, fd_set*, fd_set*, timespec const*) + 596
2   org.qt-project.QtCore         	0x000000010182c000 QEventDispatcherUNIXPrivate::doSelect(QFlags<QEventLoop::ProcessEventsFlag>, timespec*) + 672
3   org.qt-project.QtCore         	0x000000010182d0ca QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 234
4   org.qt-project.QtCore         	0x00000001017d75ad QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381
5   org.qt-project.QtCore         	0x00000001015cc905 QThread::exec() + 117
6   org.qt-project.QtCore         	0x00000001015d0393 QThreadPrivate::start(void*) + 339
7   libsystem_pthread.dylib       	0x00007fff8df6e899 _pthread_body + 138
8   libsystem_pthread.dylib       	0x00007fff8df6e72a _pthread_start + 137
9   libsystem_pthread.dylib       	0x00007fff8df72fc9 thread_start + 13

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0xc000000000000000  rbx: 0x000000010cfdfc80  rcx: 0x000000000000000f  rdx: 0x0000000000000006
  rdi: 0x000000010cfdfc80  rsi: 0x000000010cfdfc80  rbp: 0x00007fff5fbfdbd0  rsp: 0x00007fff5fbfdbb0
   r8: 0x0000000000000040   r9: 0x00007fff5fbfd700  r10: 0x00007fff8ef37164  r11: 0x0000000000000207
  r12: 0x000000010420b420  r13: 0x0000000104102520  r14: 0x0000000103751d10  r15: 0x0000000104102501
  rip: 0x00000001000fc426  rfl: 0x0000000000010202  cr2: 0x0000000170b11000
  
Logical CPU:     0
Error Code:      0x00000000
Trap Number:     13
```

DonLakeFlyer, 2015-04-19:
There are tons of crashes associated with re-connecting after disconnecting. This is the infamous zombie disconnected UAS problem. This one doesn't quite look like that, but if you get past it you'll hit the next one. I haven't been trying to fix them because it means reworking quite a fix of the widgets. Seemed like a waste since all the widgets needs to be converted to qml anyway which will then fix the problem.

DonLakeFlyer, 2015-04-19:
Just FYI: I've actually jumped in a couple times to fix various widgets to this problem got a little in and then bailed because it would require some big rework. The crashes have been around for quite some time. Any opinion on what to do? Any fix to the existing code and/or convert to Qml will take quite a while. So we are kind of stuck with it for a while either way. We could disable reconnect, and warn the user they need to restart to at least prevent the crashes. I would tend toward that and then put some focus to converting widgets to Qml.

DonLakeFlyer, 2015-04-20:
My above statement about this being around for a while is incorrect. The code bug has been around, but the crashes have not. It is showing up more now because a disconnect now deletes the UAS.

DonLakeFlyer, 2015-04-20:
Although what I say above is all nice. It's completely wrong with respect to this bug. This is one of those '"How the hell this that ever work bugs" in airframe config. Working on it.

LorenzMeier, 2015-04-20:
I can understand all the reasoning behind what you said, I have just two comments: Do we need to delete the UAS object if it causes so much instability? Can we just keep it around for now? And the other thing is: Isn't an app crash a straight no-no? If we think it'll take time to convert everything I think we need a mitigation strategy in the meantime.

DonLakeFlyer, 2015-04-20:
> Do we need to delete the UAS object if it causes so much instability

Just trading off one problem for another. If you keep the UAS around and the user tries to talk with it it's going to go boom because it no longer has a connection. So the problem just shows up a little later on. 

Yes crash is bad. That's why I was asking whether for now we should disallow re-connect until I can fix the widgets. I'll take a look again at this.

LorenzMeier, 2015-04-20:
I think disabling anything if it prevents a crash sounds fair 8).

DonLakeFlyer, 2015-04-20:
I'll do that as a last resort. First I'll try to figure out some quicker hacks as temp solution till qml convert.

closed 2015-04-20 by DonLakeFlyer