mavlink/qgroundcontrol #1495: Allow the user to configure OSDs / companion links via QML

LorenzMeier, 2015-04-23:
My apologies in advance for filing a feature request as issue, I just don't have a better approach right now to map this, and its a quick fix: We have more users wanting to connect an OSD or companion computer. The SYS_COMPANION parameter currently supports two values: 921600 and 57600 baud. It defaults to 0, if its either of these two baud rates it will launch a 2nd MAVLink instance on the TELEM2 port.

With this I want to raise the general "peripheral config" question. Is this something we want to defer, or are there more things like this we consider worthwhile enough to build a QML GUI soon?

LorenzMeier, 2015-04-23:
@DonLakeFlyer @dogmaphobic Would be great to have your general feedback here - I'm worried about sidetracking the general effort / big picture with requests like this one, and I'm unsure where I should keep them as notes. I really should just be sending PR's, but I'm flat out currently.

dogmaphobic, 2015-04-23:
I didn't even know it supported 57600 directly :) The back end is somewhat simple. Accept any (sensible) value and make sure both ends understand them. The front end however is anybody's guess. Right now we have an unwieldy *Parameter Editor* dialog with every possible parameter. It's better than nothing but it's daunting to say the least. What are you suggesting? Another entry in the *Setup*? What else could be grouped into this so we don't have a whole page for one single entry? As it is, we already have more Setup entries that will fit on the screen and it doesn't scroll.

LorenzMeier, 2015-04-23:
The last point is what I'm raising yes.

DonLakeFlyer, 2015-04-23:
I would include it with the other list of optional hardware that we currently don't deal with: OSD, Radio, Servo driven Gimbal and so forth. Mission Planner has a whole section for these in Setup. Maybe for now just clean up the parameter meta data which is not very descriptive. Then add simple search capability to parameter editor. Long Term we need to add the rest of optional hardware stuff.

[image]

LorenzMeier, 2015-04-24:
Fair enough. Lets fix meta data and docs, and point users to it. Doing a better job in the param system is probable better than implementing 100'000 new config UI entries for the 1-5% use cases.

FPVflyer, 2015-04-24:
Hi Don, pls can you paste here what to put in the txt file to enable 57600 baud rate?

+1 for the parameter description into the v2.4.0 QGC

DonLakeFlyer, 2015-04-24:
As far as I can tell 57600 is not currently supported. The mavlink start command is init.d:

	#
	# MAVLink onboard / TELEM2
	#
	if ver hwcmp PX4FMU_V2
	then
		if param compare SYS_COMPANION 921600
		then
			mavlink start -d /dev/ttyS2 -b 921600 -m onboard -r 20000
		fi
	fi

I guess you could modify that to support 57600 as well.

LorenzMeier, 2015-04-24:
I think you're looking at old code.
https://github.com/PX4/Firmware/blob/d5a9dffd43b699dfd482fb2ef70677e759eda2af/ROMFS/px4fmu_common/init.d/rcS#L462-L474

DonLakeFlyer, 2015-04-24:
> Hi Don, pls can you paste here what to put in the txt file to enable 57600 baud rate?

Just use the latest firmware and change the parameter to 57600.

LorenzMeier, 2015-04-25:
Ok, I'm closing this since we now know how we want to move on this one.

closed 2015-04-25 by LorenzMeier