mavlink/qgroundcontrol #1504: Crash if some parameter doesn't exist

DrTon, 2015-04-26:
There are a lot of places where qgroundcontrol assumes that autopilot have some parameter or some valid values. E.g.: try to remove SYS_AUTOSTART parameter or set system ID to 0: qgc will crash immediately. Even without Q_ASSERT it will crash, because of segmentation fault. Need to add handling of "no such parameter" case for each usage (a LOT of usages!), else qgc will always work only with latest versions of PX4, while it can support any mavlink autopilot in theory. The problem is in Q_ASSERT usage and corresponding API design, that uses ideology: "something unexpected? -> crash". While "something unexpected" in most cases is NOT qgc bug, but just different autopilot version.

LorenzMeier, 2015-04-26:
@DonLakeFlyer This goes a bit back to our previous discussion about this subject and I only recall it was fixed for the last instance we hit it, but not sure how.

DonLakeFlyer, 2015-04-26:
This is just a bug with SYS_AUTOSTART=0 handling in airframe config. It's isn't related to a parameter missing problem. In this case it is doing what an assert should do which is indicate a bug in the code logic. 

QGC is not coded defensively in all cases. For example if a memory allocation fails it will assert in debug and crash in release. The issue with parameters suddenly not in existence and coding defensively for that is just like you said: "Need to add handling of "no such parameter" case for each usage (a LOT of usages!)". You are talking about hundreds of convoluted error paths. Take SYS_AUTOSTART as an example. If that doesn't exist what exactly do you do? Best you can do is pop and error message and immediately shut down (which is a good idea and I will change QGC to do that).

This is the same as saying that every call to param_find by the firmware must be checked to see if it succeeded. If you look at the firmware there are no checks for param_find failing. This is exactly the same issue as QGC. The firmware expects parameters to exist and not disappear. QGC in the same way expects a core set of parameters (not all) to always exist. That expectation allows you to not have to write 10 times as much error checking code then real code.

DonLakeFlyer, 2015-04-26:
> else qgc will always work only with latest versions of PX4, while it can support any mavlink autopilot in theory.

Just to be clear. The setup screens are tied directly to a specific firmware stack and hence tied directly to a parameter set. They are not generic to a mavlink based autopilot. If you connect to a non px4fmu firmware stack with QGC (like APM) you don't get setup. The majority of QGC is mavlink generic, but setup is not.

DrTon, 2015-04-26:
>If you connect to a non px4fmu firmware stack with QGC (like APM) you don't get setup

It's ok for me, I can adjust parameters by hands, but I don't want it to crash!

DrTon, 2015-04-26:
>This is just a bug with SYS_AUTOSTART=0 handling in airframe config. It's isn't related to a parameter missing problem.

It's not only one bug, look:

- https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/PX4/PowerComponent.cc#L61
- https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/PX4/RadioComponent.cc#L65
- https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/PX4/SensorsComponent.cc#L65
- ... and a lot of other places

Just imagine that there is no such parameter (e.g. changed in new version of PX4). In debug build this assertion will fail:
https://github.com/mavlink/qgroundcontrol/blob/master/src/FactSystem/ParameterLoader.cc#L272
In release segmentation fault will happen when doing (Fact *) nullptr->value()

And I just found my fail: I set autopilot type to PX4, with GENERIC it's works better :) But anyway this approach with strict assumptions about parameters is not good, from time to time qgc will crash if versions doesn't match. IMO ground station should NEVER crash because of bad incoming data.

DonLakeFlyer, 2015-04-26:
> from time to time qgc will crash if versions doesn't match

I'm going to change it so it won't crash. But there is no way to reasonable way to make all the QGC UI continue to work if parameters that it needs is missing.

DrTon, 2015-04-26:
Ok! I understand, but "You are using incompatible version, this feature will be not available" message is better than crash :)

DonLakeFlyer, 2015-04-26:
> "You are using incompatible version, this feature will be not available" message is better than crash :)

Working on it

dogmaphobic, 2015-04-27:
I'm now utterly confused with the state of things. I just updated the firmware to the current *Stable* and running QGC current master. All sorts of fireworks go off. What firmware should I be using when using master QGC?

This comes out on the console:
```
silenceTimeout 350
TYPE CHANGED TO: 13
Add new UAS:  1
Unknown element in XML:  "board"
Parameter meta data with bad default value, name: "MAV_TYPE"  type: "INT32"  default: "MAV_TYPE_FIXED_WING"
Duplicate parameter found: "MC_ROLL_P"
Duplicate parameter found: "MC_ROLLRATE_P"
Duplicate parameter found: "MC_ROLLRATE_I"
Duplicate parameter found: "MC_ROLLRATE_D"
Duplicate parameter found: "MC_PITCH_P"
Duplicate parameter found: "MC_PITCHRATE_P"
Duplicate parameter found: "MC_PITCHRATE_I"
Duplicate parameter found: "MC_PITCHRATE_D"
Duplicate parameter found: "MC_YAW_P"
Duplicate parameter found: "MC_YAWRATE_P"
Duplicate parameter found: "MC_YAWRATE_I"
Duplicate parameter found: "MC_YAWRATE_D"
Duplicate parameter found: "MC_YAW_FF"
Duplicate parameter found: "MC_YAWRATE_MAX"
Duplicate parameter found: "MC_ACRO_R_MAX"
Duplicate parameter found: "MC_ACRO_P_MAX"
Duplicate parameter found: "MC_ACRO_Y_MAX"
Parameter meta data with bad default value, name: "MPC_MAN_R_MAX"  type: "FLOAT"  default: ""
Parameter meta data with bad default value, name: "MPC_MAN_P_MAX"  type: "FLOAT"  default: ""
Parameter meta data with bad default value, name: "MPC_MAN_Y_MAX"  type: "FLOAT"  default: ""
Duplicate parameter found: "MPC_MAN_R_MAX"
Duplicate parameter found: "MPC_MAN_P_MAX"
Duplicate parameter found: "MPC_MAN_Y_MAX"
Duplicate parameter found: "MPC_THR_MIN"
Duplicate parameter found: "MPC_THR_MAX"
Duplicate parameter found: "MPC_Z_P"
Duplicate parameter found: "MPC_Z_VEL_P"
Duplicate parameter found: "MPC_Z_VEL_I"
Duplicate parameter found: "MPC_Z_VEL_D"
Duplicate parameter found: "MPC_Z_VEL_MAX"
Duplicate parameter found: "MPC_Z_FF"
Duplicate parameter found: "MPC_XY_P"
Duplicate parameter found: "MPC_XY_VEL_P"
Duplicate parameter found: "MPC_XY_VEL_I"
Duplicate parameter found: "MPC_XY_VEL_D"
Duplicate parameter found: "MPC_XY_VEL_MAX"
Duplicate parameter found: "MPC_XY_FF"
Duplicate parameter found: "MPC_TILTMAX_AIR"
Duplicate parameter found: "MPC_TILTMAX_LND"
Duplicate parameter found: "MPC_LAND_SPEED"
Parameter meta data with bad default value, name: "SUB_INTERV"  type: "INT32"  default: ""
Parameter meta data with bad default value, name: "SUB_TESTF"  type: "FLOAT"  default: ""
```

And when I go to the *Sensors* tab, I get this:

[image]

DonLakeFlyer, 2015-04-27:
Use very latest master.

DonLakeFlyer, 2015-04-27:
I also have all the qDebug crap fixed in a pull I’m putting together now

DonLakeFlyer, 2015-04-27:
Very latest firmware master has fix for Panic Shutdown in it.

dogmaphobic, 2015-04-27:
Is there a way to keep track of firmware version and alert the user they need to upgrade the firmware before using QGC?

DonLakeFlyer, 2015-04-27:
Lorenz and I discussing blacklist support for firmware versions. That said you need to connect to detect firmware version.

dogmaphobic, 2015-04-27:
I know that but once a connection is made a check could be done before allowing you to do anything.

DonLakeFlyer, 2015-04-27:
That's the blacklist thing:
- QGC pulls down latest blacklist from web at boot
- Checks firmware rev against blacklist at connect

NYI

DonLakeFlyer, 2015-04-27:
FYI: I may be wrong about your specific case of Panic Shutdown. I"ll have to look at how the code used CAL_MAG2_ID. Not sure if it is required or not.

DonLakeFlyer, 2015-04-27:
Nope, I was wrong. Just checked firmware source. Latest master should work fine.

@LorenzMeier Looks like Stable has the parameter list problems in it. May need to update?

DonLakeFlyer, 2015-04-27:
@dogmaphobic To fill you in on what is going on here: 

A few weeks back the firmware was modified to only send parameters which have been referenced by code at boot when you do list request . This was done to cut down on initial parameter load times. 

The problem is that QGC cannot run when the parameters it needs for ui are missing. There just isn't any reasonable way to code the ui defensively again any random parameter suddenly going missing in all cases.

Previously the FactSystem would just assert in this case because it tended to indicate a QGC bug since the full list of parameters used to always be there. With the firmware change that isn't necessarily the case any more. Hence the whole Panic Shutdown thing. When QGC hits the case of a missing parameter there isn't really anything it can do other than bail out.

We've been working through a set of bugs on the Firmware side for a few days now to cover the cases which were causing QGC problem with the new param mechanism. We think we have them all fixed now.

Lorenz and I have been discussing alternate firmware implementations which would cut down param load times, as well as be more stable with respect to required parameters not suddenly disappearing.

closed 2015-04-27 by DonLakeFlyer

DonLakeFlyer, 2015-04-27:
Also if you have any better ideas how to handle missing parameters, throw 'em out there.