mavlink/qgroundcontrol #1507: Ability to cancel parameter change

AndreasAntener, 2015-04-27:
(QGC master from today)
ATM it doesn't seem to be possible to abort a parameter change. Since the changes are now transmitted instantly, it should at least be possible to hit [ESC] to revert an accidental/unwanted change.

LorenzMeier, 2015-04-27:
Actually that's a very important point - the send safeguard is missing now.

@DonLakeFlyer it would be nice if the UI would show very boldly it's going to send the values on hitting enter or leaving the field. You can easily crash the vehicle with a wrong param, and at the level of the param editor we can't prevent that, unless we introduce a special mode:

One example: the param editor has to allow a roll P gain of zero, because that's how you can best tune the rate gains. Doing that in hover will lead to a crash.

Now we have two options: do the field validation feedback-only, so the field turns red if you enter zero. Or add a checkbox to disengage field validation and only accept safe values if validation is on. What did you have in mind?

At any rate, field reset on ESC sounds like a real need.

DonLakeFlyer, 2015-04-27:
ESC is doable. Validation against the meta data is going to be added. But in the case you describe if 0 is valid as far as the meta data is concerned then it will be allowed even once validation is implemented. If the meta data says 0 is invalid it won't be allowed unless we add some sort of mode which lets you override the validation. That last option would seem to be safer.

DonLakeFlyer, 2015-04-27:
Another option: Be much more annoying when changing parameters while armed. In this case add an "Are you sure?" with warning about crashing vehicle. I kind of like that option.

AndreasAntener, 2015-04-27:
@DonLakeFlyer : no please don't make it more annoying ;) especially not with more dialogs that ask stuff all the times..
If the GUI stays like it is now, ESC ability is a must.

Regarding validation: I'm not sure why the old "changed before send" view disappeared, but validation messages or tips/hints could have easily been displayed there. So the user gets a chance to review his changes, and maybe override validation errors.

DonLakeFlyer, 2015-04-27:
Old style is gone because I hand't figured out how to implement it with the new FactSystem architecture. I think I have an idea on how to bring it back.

DonLakeFlyer, 2015-04-27:
I can't seem to get Escape to work, still trying. But Undo does work for me at least on OSX. Command+Z. Probably works on other OS as well.

AndreasAntener, 2015-04-28:
Hm.. if you can get the old pre-send view back, ESC doesn't need to work there anymore of course ;)
Just an idea: Since we have a nice parameter description column now, maybe it makes sense to split the view in 2 rows. The bottom row would then be the old pre-send view. This would also work better in the analyze perspective.

DonLakeFlyer, 2015-05-10:
All different now

closed 2015-05-10 by DonLakeFlyer