mavlink/qgroundcontrol #1508: panic shutdown on connecting to PX4Flow

kd0aij, 2015-04-27:
complains there is no SYS_AUTOSTART parameter and segfaults

DonLakeFlyer, 2015-04-27:
What do you means connecting to PX4Flow? You are connecting QGC to a PX4Flow directly? If so, to do what?

kd0aij, 2015-04-27:
That's how I used to focus the camera

kd0aij, 2015-04-27:
https://pixhawk.org/modules/px4flow#image_quality_and_output

DonLakeFlyer, 2015-04-27:
Ok. Thanks. I'll have to look but it seems like the PX4Flow is incorrectly reporting itself as a Pixhawk with the px4fmu stack running on it.

@LorenzMeier I haven't debugged but it looks like the flow board is incorrectly reporting itself as a MAV_AUTOPILOT_PX4 in the heartbeat message. This causes the autopilot plugin system to think it's a Pixhawk running px4fmu stack. I've never delved into the flow source. If you have, maybe quick check to see what's going on here. It needs to report itself as MAV_AUTOPILOT_GENERIC since their doesn't seem to be a MAV_AUTOPILOT entry for flow.

DonLakeFlyer, 2015-04-27:
Lorenz is fixing this

LorenzMeier, 2015-05-12:
Fixed.

closed 2015-05-12 by LorenzMeier