mavlink/qgroundcontrol #1522: Test Report

tumbili, 2015-04-29:
@DonLakeFlyer  Hi Don

I was doing a vehicle setup session with a person who was completely new to QGroundControl.
In general everything went really well but still there are some things I should mention:

1) I could connect the vehicle via telemetry but I could not start the setup process because the bar on the left side was not visible.
So I had to connect via USB to start the setup. We were able to do the sensor calibration via telemetry but only after we have finished the previous steps via USB.

2) When selecting the proper airframe the test person clicked on the corresponding name of the vehicle and assumed he was done. He then clicked on the save and restart button but realized that the airframe type was not set. He needed to mark the check box for the airframe type to be stored. Is it really necessary to have this check box?
Also always after clicking the save and restart button QGroundControl crashed.

3) Radio calibration went really well. The text for the instructions should be much larger, there is a lot of space left in the window which should be used.
Also maybe the buttons can be made larger.

4) The sensor calibration process is getting pretty good. The person was not sure if the direction of rotation for the mag calibration matters. So maybe just before the mag calibration starts a box could pop up which gives a brief explanation of what the user has to do. The user can then confirm that he has understood and the process can begin.

DonLakeFlyer, 2015-04-29:
We're you running the very latest firmware and QGC?

#1) Connecting over telemetry was fixed a couple days ago in firmware.
#2) Crashing and airframe config restart was fixed a couple days ago in QGC. You may still be hitting a new problem here through.
#3) RC Cal needs to be converted to Qml I'll redo then.
#4) Rotation direction doesn't actually matter. That was just the best arrow I could find on the internet! I'll look at way to improve instructions.

tumbili, 2015-04-29:
Yes

DonLakeFlyer, 2015-04-30:
I tried to repro #1 but can't get it to fail over telemetry. I know I fixed a bug there a couple days ago. Can you retry with last nights daily build? Also what OS?

tumbili, 2015-04-30:
@DonLakeFlyer Yeah will do it tomorrow. I'm using Ubuntu 14.04

DonLakeFlyer, 2015-04-30:
I retried #2 as well both over USB and telemetry and it was fine. Could be an Ubuntu only problem.

closed 2015-10-25 by LorenzMeier