mavlink/qgroundcontrol #1525: Can't download QGroundControl-Installer-Win32-Stable-V2.3.0.exe

spacemig, 2015-04-29:
I've been trying to download the latest QGroundControl for a couple of days 
http://qgroundcontrol.org/downloads
and I keep getting an error (not sure if this is the case but it seems one of the routers in my university network detects there is a virus or malware in the .exe)

Also I can't get the file from 
http://qgroundcontrol.org/oldstable
(the link is broken)

Any other places I can get QGroundControl from?

DonLakeFlyer, 2015-04-29:
I deleted the older stable version by mistake a while ago. I’ve heard the malware thing before. Not sure what I can do about that since as far as I know there is no problem with them. They are hosted on AWS.

DonLakeFlyer, 2015-05-10:
Not much I can do if you are blocked from the download location. Majority of users have no issue with this location (Amazon Web Services). Only suggestion is to download from some other internet connection that doesn't block it.

closed 2015-05-10 by DonLakeFlyer