mavlink/qgroundcontrol #1526: Params don't show up in QGroundControl

tumbili, 2015-04-30:
Wanted to tune the attitude controller today via telemetry but QGroundControl did not show all of the tuning parameters. Actually only the feed-forward gains were shown, the rest didn't appear.

DonLakeFlyer, 2015-04-30:
QGC is going to show all the parameters it gets sent. If parameter load completed:
- The progress bar should have gone all the way to the right and then gone away
- Setup screens are available
Then as far as QGC is concerned according to the mavlink protocol it has all the parameters the firmware is advertising.

If the above happened then this is likely a problem with the optional parameter support in firmware. Optional parameters can cause some parameters to not be sent to QQC depending on how the firmware code is written. So in that case the bug is on the firmware side.

DonLakeFlyer, 2015-05-07:
@tumbili If this is still happening can you move this bug to the firmware side and list the missing parameters.

closed 2015-05-07 by DonLakeFlyer