mavlink/qgroundcontrol #1527: Panic Shutdown without parameters

boosfelm, 2015-04-30:
Hi @DonLakeFlyer 
@surberj and I work with qgc and slightly modified px4 version from Tuesday (april 28th) and when I tried to connect to qgc with a flashed firmware but no parameters, I got the following warning and right after it a panic shutdown.
[image]
[image]

I managed to bypass this by using a qgc 2.1, uploading an old parameter file and building up on that. With the new qgc is was impossible to continue otherwise and from user-perspective you feel somewhat helpless, because of gqcs drastic response ;)

Any thoughts on that? Anyways, thanks for the huge effort improving qgc in general - great work!

DonLakeFlyer, 2015-04-30:
QGC requires the parameters be available in order to run. When you say flash firmware but no parameters, how did you end up with no parameters?

boosfelm, 2015-05-04:
Hi @DonLakeFlyer 
I bascially flashed a new firmware and powered the Pixhawk with a startup script that would fetch parameter from sd_card, but has none on it. This way it should then just stick to the default values from build and launch qgc, right? However qgc does not launch and exits with the panic shutdown. Can you point me to the problem? Do I have to change anything in my mavlink protocol if I only add new parameters?

DonLakeFlyer, 2015-05-04:
Not sure what you are trying to accomplish. Do you want to write your our custom version of the firmware? Mavlink protocol doesn't really have anything to do with parameters other than providing a mechanism to set/get.

The firmware will only send parameters which have been referenced by a param_find during boot to QGC. If you've modified the firmware in such a way that SYS_AUTOSTART is no longer referenced at boot then QGC will not run. 

If you modified the firmware in such a way that you have removed some of the parameters from the firmware, or changed the behavior of the firmware in such as way that it is no longer compliant with how the px4fmu stack runs, then you should not report yourself as a MAV_AUTOPILOT_PX4 in the heartbeat. You should report yourself as MAV_AUTOPILOT_GENERIC. Newly added parameters are fine.

DonLakeFlyer, 2015-05-06:
This is all different now: Pull #1537. That said if the parameters don't come across there isn't much QGC can do about it. There is a problem on the firmware side.

closed 2015-05-06 by DonLakeFlyer