mavlink/qgroundcontrol #1528: No setup options with worng io

boosfelm, 2015-04-30:
Hi
I tried to connect a new px4 firmware with a wrong io update to qgc. It is actually possible and in the fly tab I can even see the estimator running. However, it is impossible to see the full setup options and the hint to the parameter list is somewhat misleading.
Can I or @surberj help somehow?

[image]

DonLakeFlyer, 2015-04-30:
There should be a progress indicator that shows up right below the toolbar when you connect to a vehicle. Did that start moving to the right and stop somewhere in the middle? Or did it never show up at all. Are you connecting over telemetry? The setup options will only show up after the full set of parameters is downloaded.

boosfelm, 2015-05-04:
Hi @DonLakeFlyer 
The progress bar did not show up at all. I try telemetry and ftdi, both times the same behaviour. It really seems there is something wrong with the parameter, but I don't quiet get where it breaks.

boosfelm, 2015-05-04:
I investigated a little further and it is really weird that all functionalities work except the setup tab. Any idea what is wrong?
[image]
[image]
[image]

DonLakeFlyer, 2015-05-04:
> connect a new px4 firmware with a wrong io

Do you mean that the board is running the wrong firmware? If so what firmware is running?

boosfelm, 2015-05-10:
I believe this was also due to a mess in the parameters, never happened again in the later qgc commit. Thanks for your help.

closed 2015-05-10 by boosfelm