mavlink/qgroundcontrol #1533: Offline editing of Waypoints

ChauvinLou, 2015-05-03:
Latest version of QGroundController appears to require that you are connected to a device prior to being able to create a waypoint file.

DonLakeFlyer, 2015-05-04:
Works fine for me. But then it's next to impossible to understand what you tried to do and what failed, given the little detail you have provided.

ChauvinLou, 2015-05-04:
Agreed - here is additional information:

Windows Version 2.3.0 Stable
Windows 7 Professional Service Pack 1
4 GB Memory
64-bit OS
Vehicle not powered on
Start QGround Controller.
Load waypoints from Waypoint file (8 waypoints. launch, 6 waypoints, land) 
Waypoints load ok
Waypoints do not display on map (as they do when connected to vehicle)
Double Clicking on Map after selecting Last Pos - does nothing.
Add waypoint from the table selection works, but doesn't display the waypoint on the map

DonLakeFlyer, 2015-05-05:
@dogmaphobic Could you take a look at this one?

dogmaphobic, 2015-05-06:
Has it ever been any different? I just tried an older version from last December and it did the same. As far as I can tell, the original plan view only deals with waypoints once it gets them from the vehicle. I'm not sure it's worth even looking at fixing that if we are to replace it all.

DonLakeFlyer, 2015-05-06:
Don't know. If that's the way it always was doesn't make sense to add the feature now.

Susurrus, 2015-05-06:
It's actually a feature I've been dying for and @DonLakeFlyer I don't see how this feature not being implemented is reason to never implement it.

I like to plan out missions well before hand and I don't always have my autopilot around to plan them, like if I'm at home or the AP is outside. It would definitely be useful to have this functionality to make pre-planning missions easier. There's definitely no technical nor logical reason for this functionality not to exist. I would also think it'd be helpful for new users to try out this functionality without needing an autopilot so we can "sell" them on QGC before they have all the necessary hardware.

dogmaphobic, 2015-05-06:
It only means there is no point in implementing it as part of the old Mission Planning. It will be in the new one.

DonLakeFlyer, 2015-05-06:
Closing this since it's not a bug.

closed 2015-05-06 by DonLakeFlyer