mavlink/qgroundcontrol #1536: More "direct" parameter editing

AndreasAntener, 2015-05-04:
Hey @DonLakeFlyer , the new parameter editor looks cool!

I have some suggestions as a heavy parameter changer:
- if I click on a parameter, it would be nice if the focus is already in the text field AND the text is selected. (the focus is there once you modified one parameter but is lost after doing other stuff)
- [ENTER] to save
- [ESC] to cancel

DonLakeFlyer, 2015-05-04:
This is all different in the latest daily build. Try that out.

AndreasAntener, 2015-05-04:
I just compiled it from master

DonLakeFlyer, 2015-05-04:
Sorry I was thinking the wrong thing! I think this should all be doable.

DonLakeFlyer, 2015-05-25:
Tried getting this to work but couldn't. I try again at some point.

AndreasAntener, 2015-05-26:
Ok, thanks for fixing all the other issues :). I noticed that when I press ESC the window gets smaller and the info view disappears, strange..

AndreasAntener, 2015-07-15:
@DonLakeFlyer Awesome! just saw in the user list that you were able to fix this :+1:

closed 2015-07-15 by AndreasAntener