mavlink/qgroundcontrol #1538: 3rd radio telemetry doesnt work anymore with newest QGC and newest firmware

severinleuenberger, 2015-05-06:
with the newes px4 firmware (flashed 60 minutes ago) combined with the newest QGC (20 min ago) , the telemetry-bind doesnt work anymore. It hangs completely. Tried it 20 times:
not luck.

 [I] at :0 - "waitingReadParamIndexCount: 21"
[I] at :0 - "waitingParamCount: 21"
[I] at :0 - "Received PARAM_VALUE "NAV_AH_LON" QVariant(int, 1518423250) 6"
[I] at :0 - "_parameterUpdate (usaId: 107 componentId: 50 name: "NAV_AH_LON" cou
nt: 362 index: 153 mavType: 6 value: QVariant(int, 1518423250) )"
[I] at :0 - "_waitingReadParamIndexMap: (60, 85, 86, 95, 96, 208, 209, 216, 217,
235, 236, 245, 246, 271, 272, 311, 312, 321, 322, 354, 355)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 21"
[I] at :0 - "waitingParamCount: 21"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 60 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 85 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 86 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 95 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 96 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 208 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 209 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 216 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 217 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 235 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 236 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 60 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 85 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 86 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 95 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 96 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 208 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 209 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 216 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 217 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 235 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 236 )"
[I] at :0 - "Received PARAM_VALUE "PE_MAGB_PNOISE" QVariant(float, 0.0003) 9"
[I] at :0 - "_parameterUpdate (usaId: 107 componentId: 50 name: "PE_MAGB_PNOISE"
count: 362 index: 24 mavType: 9 value: QVariant(float, 0.0003) )"
[I] at :0 - "_waitingReadParamIndexMap: (60, 85, 86, 95, 96, 208, 209, 216, 217,
235, 236, 245, 246, 271, 272, 311, 312, 321, 322, 354, 355)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 21"
[I] at :0 - "waitingParamCount: 21"
[I] at :0 - "Received PARAM_VALUE "FW_YCO_METHOD" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 107 componentId: 50 name: "FW_YCO_METHOD"
count: 362 index: 48 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: (60, 85, 86, 95, 96, 208, 209, 216, 217,
235, 236, 245, 246, 271, 272, 311, 312, 321, 322, 354, 355)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 21"
[I] at :0 - "waitingParamCount: 21"
[I] at :0 - "Received PARAM_VALUE "FW_YCO_VMIN" QVariant(float, 9) 9"
[I] at :0 - "_parameterUpdate (usaId: 107 componentId: 50 name: "FW_YCO_VMIN" co
unt: 362 index: 49 mavType: 9 value: QVariant(float, 9) )"
[I] at :0 - "_waitingReadParamIndexMap: (60, 85, 86, 95, 96, 208, 209, 216, 217,
235, 236, 245, 246, 271, 272, 311, 312, 321, 322, 354, 355)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 21"
[I] at :0 - "waitingParamCount: 21"
[I] at :0 - "Received PARAM_VALUE "FW_RR_IMAX" QVariant(float, 0.2) 9"
[I] at :0 - "_parameterUpdate (usaId: 107 componentId: 50 name: "FW_RR_IMAX" cou
nt: 362 index: 58 mavType: 9 value: QVariant(float, 0.2) )"
[I] at :0 - "_waitingReadParamIndexMap: (60, 85, 86, 95, 96, 208, 209, 216, 217,
235, 236, 245, 246, 271, 272, 311, 312, 321, 322, 354, 355)"
[I] at :0 - "_waitingReadParamNameMap

standard 3rd 433er radio used, worked perfect for month now, but today it seems be broken

severinleuenberger, 2015-05-06:
I have 3 machines running with 3 versions of QGC. With an older version of QGC i can connect with px4 trough 433 3rd radio. but there i cannot edit parameters anymore. An errormessage apperars  (incompatibility between QGC and PX4 Firmware) and then quits.

LorenzMeier, 2015-05-06:
Does it work via USB?

severinleuenberger, 2015-05-06:
Yes!! Just tried it with USB. There it works. But not on standard 3rd radio.

severinleuenberger, 2015-05-06:
Ooops, but when I try then to calibrate sensors, the following message appears and shut down the QGC: 

[image]

LorenzMeier, 2015-05-06:
Your firmware is definitely on master, and the firmware boot didn't fail? Because that parameter is present , I just cross-checked the code. This could only happen if a) the sensors app didn't start (which would indicate a failure in the boot process very early on) or b) the param transmission is also flaky via USB.

severinleuenberger, 2015-05-06:
I did via console: 
git fetch --all 
make distclean 
make archives
make 
make upload px4fmu-v2_default // upload starten

Did I missed a step?

LorenzMeier, 2015-05-06:
Git fetch does not actually update the files in your working copy. You need to either run git fetch && ```git merge upstream/master``` or ```git pull upstream/master``` (which is an alias for these commands).

severinleuenberger, 2015-05-06:
Ok. Shame on me. Thanks!

LorenzMeier, 2015-05-06:
Did you verify its fixed for you now?

severinleuenberger, 2015-05-06:
I just do merge now and will report back, if it is a problem. 
For now, I think I made a mistake.

severinleuenberger, 2015-05-06:
...I do not get it merged...
[image]
Missed something again?

severinleuenberger, 2015-05-06:
Meine Einstellungen auf git sollten eingentlich stimmen....

$ git config --list
core.repositoryformatversion=0
core.filemode=false
core.bare=false
core.logallrefupdates=true
core.symlinks=false
core.ignorecase=true
core.hidedotfiles=dotGitOnly
remote.origin.fetch=+refs/heads/*:refs/remotes/origin/*
remote.origin.url=https://github.com/PX4/Firmware.git
branch.master.remote=origin
branch.master.merge=refs/heads/master
submodule.NuttX.url=git://github.com/PX4/NuttX.git
submodule.Tools/gencpp.url=https://github.com/ros/gencpp.git
submodule.Tools/genmsg.url=https://github.com/ros/genmsg.git
submodule.mavlink/include/mavlink/v1.0.url=git://github.com/mavlink/c_library.gi
t
submodule.src/lib/eigen.url=https://github.com/PX4/eigen.git
submodule.uavcan.url=git://github.com/pavel-kirienko/uavcan.git
submodule.unittests/gtest.url=https://github.com/sjwilks/gtest.git

LorenzMeier, 2015-05-06:
In this case the command is ```git merge origin/master```. Normally you're however advised to work in your own fork and to push your changes to it.

https://help.github.com/articles/fork-a-repo/

and then:
https://help.github.com/articles/syncing-a-fork/

severinleuenberger, 2015-05-06:
Great, thanks for help. I am just running the git-commands. Let you know.

severinleuenberger, 2015-05-06:
It was completely my lack of knowledge. Now everything works. It was an old build, not the newest from git server. Thanks for your help. 
Best, Severin

closed 2015-05-06 by severinleuenberger