mavlink/qgroundcontrol #1539: Remove assertion for unknown airframe type

AndreasAntener, 2015-05-06:
The assertion for finding a SYS_AUTOSTART match makes debugging QGC hard if we're using a custom airframe config.

https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/PX4/AirframeComponentController.cc#L88
IMHO you could remove the assertions there :)

DonLakeFlyer, 2015-05-06:
The reason for that is the airframe config code right now doesn't really know how to handle a custom airframe config. It's likely to do strange things, hence the assert. The bug is to support customer airframe config and remove the assert.

LorenzMeier, 2015-05-06:
I know I'm stating the obvious, but just a text box / spin box with the current number (and accepting a number) would do. It could show instead of the normal UI, along with a short text stating that custom configs can be set, but are not supported in the UI.
Or even simpler: Hide the complete pane, let the user know that he has to use the param editor and can re-enable the GUI config by setting the param to 0 / default.

AndreasAntener, 2015-05-06:
From a firmware API perspective there isn't much to handle in the airframe configuration, maybe 2 parameters? Should be fairly easy to just show the airframe as not-configured.

But I get your point, I just disagree. I know you are making a huge effort to bring QGC towards the end user, but the developers should at least still be "allowed" to use it ;). This is why it would be nice to have single functions (views) fail fast instead of the complete QGC quitting (as when there are missing parameters).

AndreasAntener, 2015-05-06:
Oh, sorry Lorenz, didn't see your post before, yes, I was thinking too, if you open that panel you'd just get a text that tells you to look in the parameters.

LorenzMeier, 2015-05-06:
Don already did that 8)
https://github.com/mavlink/qgroundcontrol/pull/1537#event-298369182

AndreasAntener, 2015-05-06:
Holy ** that looks awesome ;) will try

DonLakeFlyer, 2015-05-10:
Fixed

closed 2015-05-10 by DonLakeFlyer