mavlink/qgroundcontrol #1540: Prevent reverse lookup for IP UDP hosts

AndreasAntener, 2015-05-06:
@DonLakeFlyer 
In https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/UDPLink.cc#L299 QHostInfo implicitly does a reverse lookup for IP hosts (or a normal lookup for name hosts). Most of the time we don't have DNS servers accessible if we're testing in private networks, which means QGC always blocks a certain amount of time per host on startup and on changing the host addresses.

I looked very briefly in the API for QHostInfo but couldn't spot a way to prevent the reverse lookup for the IP-address case (which is completely unnecessary..), maybe you have an idea?

DonLakeFlyer, 2015-05-06:
I know next to nothing about TCP/IP and DNS. So can't help much.

@dogmaphobic Gus: Any ideas?

dogmaphobic, 2015-05-06:
Yes, the idea is to get an IP address if a host name is passed. At the lower level, if you pass an IP address (instead of a host name), the function simply returns it. Qt is trying to be too smart and doing a reverse lookup. That's what's causing the DNS lookup:

*"If you pass a literal IP address to name instead of a host name, QHostInfo will search for the domain name for the IP (i.e., QHostInfo will perform a reverse lookup). On success, the returned QHostInfo will contain both the resolved domain name and IP addresses for the host name."*

I will change the code and check for an IP address myself instead of asking Qt.

dogmaphobic, 2015-05-06:
@AndreasAntener, can you please try this? (#1541) My UDP testing ability is limited to one single connection using a redirector.

closed 2015-05-06 by AndreasAntener