mavlink/qgroundcontrol #1545: raspberry pi support

TheBouwman, 2015-05-07:
There is only a linux 64x build but can you compile a arm version on your own and how?

dogmaphobic, 2015-05-07:
It's in my to do list but it's so far down there that I may not ever get to it.

Two options:

- Load the Arm tool chain and cross compile on your computer (complicated)
- Load Qt 5.4.x on the Raspberry and build it there (really slow)

Whichever you pick, once Qt is properly installed (and configured for cross compile if that's the case) you need to run qmake with the proper settings for generating the make files and then it's just a matter of running make to build it all.

Susurrus, 2015-05-07:
I know someone has done this with a BeagleBone Black, so I'd suggest bringing this up on the mailing list. They also had a nice display to go with it I think.

CodeChief, 2015-05-08:
The Emlid NAVIO+ solution (www.emlid.com) is already done and works (custom Real Time ARM Linux and APM "firmware" build available to download), similar to how the Erle Brain was done for the BBB. QGrountControl doesn't know the difference it's just like using it with any other drone with the APM firmware.

LorenzMeier, 2015-05-08:
@CodeChief You're talking about something else. This discussion is about running the GCS on a BBB-like platform. You're talking about running the autopilot on a BBB.

CodeChief, 2015-05-08:
@LorenzMeier ??? This is the QGROUNDCONTROL forum and both the question and my answer are both about RasBerry Pi. Read carefully.

closed 2015-05-08 by LorenzMeier