mavlink/qgroundcontrol #1547: Firmware upgrade usability: Asking the user to unplug is way too subtle

LorenzMeier, 2015-05-08:
The main issue is that the pop-up reminding the user about the license terms is popping up, then the error message that the board needs to be unplugged first is only in the console. This is completely counter-intuitive, as the license terms are irrelevant if the board can't be flashed.

I would suggest the following steps to resolve:

  * Don't show the license terms / notes in the pop-up, rather, show them in the widget area as foot note. That's all it takes legally.
  * If the board is plugged, ask the user via a modal pop-up to unplug it, THEN click ok. Restart the flashing process once he clicks ok

Those changes should not require much alternation of the business logic, but will substantially help to de-confuse the dialog.

DonLakeFlyer, 2015-05-10:
Firmware upgrade doesn't show license terms anywhere. Are you talking about these popups?

    if (_firmwareType == BetaFirmware) {
        warningMsg = tr("WARNING: BETA FIRMWARE\n"
                        "This firmware version is ONLY intended for beta testers. "
                        "Although it has received FLIGHT TESTING, it represents actively changed code. Do NOT use for normal operation.\n\n"
                        "Are you sure you want to continue?");
    } else if (_firmwareType == DeveloperFirmware) {
        warningMsg = tr("WARNING: CONTINUOUS BUILD FIRMWARE\n"
                        "This firmware has NOT BEEN FLIGHT TESTED. "
                        "It is only intended for DEVELOPERS. Run bench tests without props first. "
                        "Do NOT fly this without addional safety precautions. Follow the mailing "
                        "list actively when using it.\n\n"
                        "Are you sure you want to continue?");
    }

These are the scare people away from anything other than Stable pop-ups. What gets shown depends on what you select. If you select Stable you get no pop-up.

DonLakeFlyer, 2015-05-10:
I can make the unplug thing more prominent by using a right-hand dialog thingy with Cancel/Upgrade buttons. Then I can put the above messages at the bottom of that dialog as well so that they are less prominent that the unplug message.

LorenzMeier, 2015-05-12:
That suggestion makes a lot of sense and would fix that issue I guess. The prominent warning below the unplug message is good enough I think. The warning has mostly a legal reason. We found that the only way to prevent users from flying beta / master over crowds of people is to hide the option (e.g. you have to press CTRL to unhide the options from the drop-down list). The reason this helps is that people that read instructions will still find that option in the docs or learn about it from the community, while people generally ignoring instructions will never accidentally find it.

One more thing: Because QGC asks people to unplug, on some OS and depending on timing they end up with a stale connection (QGC on Mac OS and Linux still believes its connected if you pull the USB cable). @dogmaphobic does your proposed fix address this, or do you have an idea how to reliably remove the link on a hardware disconnect event? I know that that's non-trivial.

dogmaphobic, 2015-05-12:
My fix was unrelated to this. It was just a casualty of the link being deleted when the physical link disappears. The issue above is a bit complicated because of the asynchronous nature of it all (and poor handling of asynchronous disconnects). It's simple to just delete the connection and start the flashing tool on its own serial connection. The problem is that the link disconnect propagates throughout the app. You would need to make sure all the signals connected to the departing link are done processing it and disconnected. That's what's causing all these crashes and zombie states. My guess is that there is a race condition somewhere with this *link disconnected* signal and its handling. Too many modules connect this signal and do unpredictable things with it. The MavLink protocol, the UAV plugin, the Debug Console, and who knows what else, all connect to these signals.

In theory you would not need to tell the user to unplug anything. You just reboot the PX4, delete the link and transfer over to the flashing tool. For that to work cleanly (assuming rebooting the PX4 is reliable), it would be a matter of going through the code and cleaning up the link disconnect signal states. Dumping the debug console was a start :)

DonLakeFlyer, 2015-05-12:
> Because QGC asks people to unplug, on some OS and depending on timing they end up with a stale connection

This is changing. Firmware upgrade now detects QGC connections first and tells you to disconnect them, not unplug the cable. Then it will check for a plugged in cable with no QGC connection and ask to unplug. I'm changing all of this sequence to be more user friendly.

DonLakeFlyer, 2015-05-25:
Done

closed 2015-05-25 by DonLakeFlyer