mavlink/qgroundcontrol #1548: Crash after clicking on "apply and restart" in airframe config

LorenzMeier, 2015-05-08:
@DonLakeFlyer Is this the rare SYS_AUTOSTART param missing case?

```
Process:               qgroundcontrol [97135]
Path:                  /Users/USER/*/qgroundcontrol.app/Contents/MacOS/qgroundcontrol
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               2.0.0 (227)
Code Type:             X86-64 (Native)
Parent Process:        Qt Creator [94993]
Responsible:           qgroundcontrol [97135]
User ID:               502

Date/Time:             2015-05-08 10:01:16.668 +0200
OS Version:            Mac OS X 10.10.2 (14C2055)
Report Version:        11
Anonymous UUID:        FFA63A60-490C-41EB-AFE8-E82F56014EB6

Time Awake Since Boot: 11000 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called
Pure virtual function called!

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff91637286 __pthread_kill + 10
1   libsystem_c.dylib             0x00007fff9a516b53 abort + 129
2   libc++abi.dylib               0x00007fff91d7aa21 abort_message + 257
3   libc++abi.dylib               0x00007fff91da142f __cxa_pure_virtual + 18
4   org.qgroundcontrol.qgroundcontrol	0x00000001000cb470 LinkConfiguration::duplicateSettings(LinkConfiguration*) + 16
5   org.qgroundcontrol.qgroundcontrol	0x000000010010b504 QGCApplication::reconnectAfterWait(int) + 132
6   org.qgroundcontrol.qgroundcontrol	0x0000000100289b6f AirframeComponentController::changeAutostart() + 703
7   org.qgroundcontrol.qgroundcontrol	0x000000010030382c AirframeComponentController::qt_metacall(QMetaObject::Call, int, void**) + 108
8   org.qt-project.QtQml          	0x0000000101d8aa4a CallMethod(QObject*, int, int, int, int*, QV8Engine*, QV4::CallData*) + 330
9   org.qt-project.QtQml          	0x0000000101d8a359 CallPrecise(QObject*, QQmlPropertyData const&, QV8Engine*, QV4::CallData*) + 601
10  org.qt-project.QtQml          	0x0000000101d89e9c QV4::QObjectMethod::callInternal(QV4::CallData*) + 3180
11  org.qt-project.QtQml          	0x0000000101d95aaf QV4::Runtime::callProperty(QV4::ExecutionContext*, QV4::String*, QV4::CallData*) + 671
12  ???                           0x000000010c2727fd 0 + 4498860029
13  org.qt-project.QtQml          	0x0000000101d45190 QV4::SimpleScriptFunction::call(QV4::Managed*, QV4::CallData*) + 528
14  org.qt-project.QtQml          	0x0000000101e2b45e QQmlJavaScriptExpression::evaluate(QQmlContextData*, QV4::ValueRef, QV4::CallData*, bool*) + 622
15  org.qt-project.QtQml          	0x0000000101dd2142 QQmlBoundSignalExpression::evaluate(void**) + 2258
16  org.qt-project.QtQml          	0x0000000101dd2b68 QQmlBoundSignal_callback(QQmlNotifierEndpoint*, void**) + 440
17  org.qt-project.QtQml          	0x0000000101e0d45c QQmlNotifier::emitNotify(QQmlNotifierEndpoint*, void**) + 92
18  org.qt-project.QtQml          	0x0000000101dae711 QQmlData::signalEmitted(QAbstractDeclarativeData*, QObject*, int, void**) + 753
19  org.qt-project.QtCore         0x0000000100e9dcb7 QMetaObject::activate(QObject*, int, int, void**) + 183
20  org.qt-project.QtQml          	0x0000000101da857f QQmlVMEMetaObject::metaCall(QMetaObject::Call, int, void**) + 511
21  org.qt-project.QtQml          	0x0000000101da8686 QQmlVMEMetaObject::metaCall(QMetaObject::Call, int, void**) + 774
22  org.qt-project.QtQml          	0x0000000101da8686 QQmlVMEMetaObject::metaCall(QMetaObject::Call, int, void**) + 774
23  org.qt-project.QtQml          	0x0000000101d8aa4a CallMethod(QObject*, int, int, int, int*, QV8Engine*, QV4::CallData*) + 330
24  org.qt-project.QtQml          	0x0000000101d8a359 CallPrecise(QObject*, QQmlPropertyData const&, QV8Engine*, QV4::CallData*) + 601
25  org.qt-project.QtQml          	0x0000000101d89e9c QV4::QObjectMethod::callInternal(QV4::CallData*) + 3180
26  org.qt-project.QtQml          	0x0000000101d95aaf QV4::Runtime::callProperty(QV4::ExecutionContext*, QV4::String*, QV4::CallData*) + 671
27  ???                           0x000000010c26647a 0 + 4498809978
28  org.qt-project.QtQml          	0x0000000101d45190 QV4::SimpleScriptFunction::call(QV4::Managed*, QV4::CallData*) + 528
29  org.qt-project.QtQml          	0x0000000101e2b45e QQmlJavaScriptExpression::evaluate(QQmlContextData*, QV4::ValueRef, QV4::CallData*, bool*) + 622
30  org.qt-project.QtQml          	0x0000000101dd2142 QQmlBoundSignalExpression::evaluate(void**) + 2258
31  org.qt-project.QtQml          	0x0000000101dd2b68 QQmlBoundSignal_callback(QQmlNotifierEndpoint*, void**) + 440
32  org.qt-project.QtQml          	0x0000000101e0d45c QQmlNotifier::emitNotify(QQmlNotifierEndpoint*, void**) + 92
33  org.qt-project.QtQml          	0x0000000101dae711 QQmlData::signalEmitted(QAbstractDeclarativeData*, QObject*, int, void**) + 753
34  org.qt-project.QtCore         0x0000000100e9dcb7 QMetaObject::activate(QObject*, int, int, void**) + 183
35  libqtquickcontrolsplugin.dylib	0x000000010d203cfd QQuickAction::triggered(QObject*) + 61
36  libqtquickcontrolsplugin.dylib	0x000000010d203c63 QQuickAction::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) + 979
37  libqtquickcontrolsplugin.dylib	0x000000010d203fb9 QQuickAction::qt_metacall(QMetaObject::Call, int, void**) + 89
38  org.qt-project.QtQml          	0x0000000101d8bea5 CallMethod(QObject*, int, int, int, int*, QV8Engine*, QV4::CallData*) + 5541
39  org.qt-project.QtQml          	0x0000000101d8a4bc CallPrecise(QObject*, QQmlPropertyData const&, QV8Engine*, QV4::CallData*) + 956
40  org.qt-project.QtQml          	0x0000000101d89e68 QV4::QObjectMethod::callInternal(QV4::CallData*) + 3128
[... 162 lines cut ...]
6   libsystem_pthread.dylib       0x00007fff92ede41d thread_start + 13

Thread 13:: QQmlThread
0   libsystem_kernel.dylib        	0x00007fff916373fa __select + 10
1   org.qt-project.QtCore         0x0000000100ebdd80 QEventDispatcherUNIXPrivate::doSelect(QFlags<QEventLoop::ProcessEventsFlag>, timespec*) + 672
2   org.qt-project.QtCore         0x0000000100ebee4a QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 234
3   org.qt-project.QtCore         0x0000000100e6937d QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381
4   org.qt-project.QtCore         0x0000000100c5f4d5 QThread::exec() + 117
5   org.qt-project.QtQml          	0x0000000101e43f09 QQmlThreadPrivate::run() + 57
6   org.qt-project.QtCore         0x0000000100c62f63 QThreadPrivate::start(void*) + 339
7   libsystem_pthread.dylib       0x00007fff92ee0268 _pthread_body + 131
8   libsystem_pthread.dylib       0x00007fff92ee01e5 _pthread_start + 176
9   libsystem_pthread.dylib       0x00007fff92ede41d thread_start + 13

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000000000006  rcx: 0x00007fff5fbf9508  rdx: 0x0000000000000000
  rdi: 0x0000000000000513  rsi: 0x0000000000000006  rbp: 0x00007fff5fbf9530  rsp: 0x00007fff5fbf9508
   r8: 0x00007fff91da37c1   r9: 0x00007fff5fbf9620  r10: 0x0000000008000000  r11: 0x0000000000000206
  r12: 0x00007fff5fbf9690  r13: 0x5e008282354b8174  r14: 0x00007fff7c192300  r15: 0x00007fff5fbf9570
  rip: 0x00007fff91637286  rfl: 0x0000000000000206  cr2: 0x00007fff7d1a6080
  
Logical CPU:     0
Error Code:      0x02000148
Trap Number:     133
```

LorenzMeier, 2015-05-08:
@dogmaphobic Since I see a link-related call in the stack you might want to have a look as well.

dogmaphobic, 2015-05-08:
I have to leave but this is what I’ve tracked down so far:

Here the PX4 is rebooted:

https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/PX4/AirframeComponentController.cc#L116

When that happens, the interface disappears and it is deleted (dynamic interfaces are not persistent):

https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/LinkManager.cc#L502-L519

Meanwhile, the reboot code is waiting for an amount of time and then it tries to reconnect using the now deleted configuration:

https://github.com/mavlink/qgroundcontrol/blob/master/src/QGCApplication.cc#L658-L673

This explains the crash. However, when I made the interface persistent (so it would not get deleted), the connection still didn’t *reconnect*. It stays in a limbo state where no data is read from the serial port but the serial link *thinks* the port is open. It’s probably safer to collect the link configuration (by name and not a pointer to an instance) before rebooting the PX4, reboot, disconnect the link altogether, wait, and start over. This *reconnect* seems way too fragile to me. I can take a shot at it unless @DonLakeFlyer wants to do something different.

Also note that on Ubuntu, the interface may very likely change the name so this *reconnect* would never work one way or the other.

LorenzMeier, 2015-05-08:
I think we should remove the auto-reconnect and just ask the user with a popup to reconnect. It should have a similar timeout though, so he doesn't try to soon.

DonLakeFlyer, 2015-05-09:
I thought I had fixed this crash a few weeks back by using duplicateSettings. The reboot of the board doesn't cause the link to go away. Just a lost connection. That will only happen when QGCApplication::reconnectAfterWait disconnects the link. It calls LinkConfiguration::duplicateSettings before the link is disconnected which creates a new free-standing settings object that QGCApplication owns. Since the usb radio or direction connection port is only disconnected from code and not physically disconnected it should not get assigned a new port number/name. That only happens when you physically disconnect. So all this should work fine.

@LorenzMeier You're using latest master right? I just tried this over a usb connection on OS X and it worked fine. How are you connected? Radio? Usb?

@dogmaphobic It crashed for you as well?

dogmaphobic, 2015-05-09:
I was using USB. Note that the crash depends on timing. It doesn't crash every time but about half the time. When you reboot the PX4, the link is lost (if you run lsusb it shows nothing connected to the USB bus). At that point, when QGC notices that the link is no longer there, it deletes the configuration (it's a dynamic, transient configuration after all). By the time you duplicate the configuration, you are copying something that has been deleted. It's all timing. It's also an issue with LinkManager deleting the LinkConfiguration (because it disappeared) while a SerialLink still holds it. Let me handle the low level first. I need to work it out so the LinkManager does something a bit more sensible when deleting disappearing links. Once I get that resolved, the crash would go away because the link configuration would still be live (while the serial link itself is still live). Then we figure out what to do. As I explained, things don't exactly go on working after a reboot, even if the link configuration is not deleted.

DonLakeFlyer, 2015-05-10:
I just need to move to reboot command until after a dup the link don't I? Easy change. Is the change to LinkManager really necessary?

DonLakeFlyer, 2015-05-10:
Ok, so I'm going to bail out on trying to reconnect since it really isn't possible to do reliably. 

@dogmaphobic Should I reboot and disconnect the link or just reboot and not worry about disconnecting the link since that will happen any way?

dogmaphobic, 2015-05-10:
In theory, once the link is gone (rebooting), the serial link will automatically disconnect. When the board shows up again, the user can use the connect button to reconnect manually. Note that I tried that here but it would only connect in one out of half a dozen tries. Normally the connection goes through but no data comes from the serial port. I had to power cycle the PX4.

DonLakeFlyer, 2015-05-10:
Ok got it. I'll just reboot.

DonLakeFlyer, 2015-05-10:
> but it would only connect in one out of half a dozen tries.

That problem has been around for a while. The board seems to get hung up if you reconnect to it. I've never been able to track down why.

closed 2015-05-25 by DonLakeFlyer