mavlink/qgroundcontrol #1550: File transfer progresses all the way to 100%, but never completes

LorenzMeier, 2015-05-09:
This is with QGC master and PX4 master:

[image]

The download seems to work fine and fast, but unfortunately it does not seem to complete.

DonLakeFlyer, 2015-05-10:
Was this over USB or radio? So far, can't repro. I'll do a code inspection next.

LorenzMeier, 2015-05-10:
Sorry for the flaky reporting - it was USB. Let me re-test again and see if I can narrow down the conditions it requires.

DonLakeFlyer, 2015-05-10:
Maybe size? I tried downloading a message file as well without problem. Also try different files to see if it's 100% repro for you or flaky.

LorenzMeier, 2015-05-10:
It's a 100% repro on a messages file with 29992 bytes size. Latest QGC and PX4 master (built just now). Other files work.

LorenzMeier, 2015-05-10:
QGC crashed after I clicked disconnect while it was in this state:

```
Process:               qgroundcontrol [9580]
Path:                  /Users/USER/*/qgroundcontrol.app/Contents/MacOS/qgroundcontrol
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               2.0.0 (227)
Code Type:             X86-64 (Native)
Parent Process:        Qt Creator [195]
Responsible:           Qt Creator [195]
User ID:               502

Date/Time:             2015-05-10 23:19:14.697 +0200
OS Version:            Mac OS X 10.10.2 (14C2055)
Report Version:        11
Anonymous UUID:        FFA63A60-490C-41EB-AFE8-E82F56014EB6

Sleep/Wake UUID:       E48C32A1-12CA-4F8A-A114-EA13FC94265E

Time Awake Since Boot: 26000 seconds
Time Since Wake:       1100 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qgroundcontrol.qgroundcontrol	0x00000001000f69d2 SerialLink::getName() const + 18
1   org.qgroundcontrol.qgroundcontrol	0x00000001001551ff DebugConsole::_setConnectionState(bool) + 831
2   org.qgroundcontrol.qgroundcontrol	0x00000001001546a1 DebugConsole::linkSelected(int) + 273
3   org.qt-project.QtCore         	0x0000000100e9e7ab QMetaObject::activate(QObject*, int, int, void**) + 2987
4   org.qt-project.QtWidgets      	0x000000010235f062 QComboBoxPrivate::_q_emitCurrentIndexChanged(QModelIndex const&) + 194
5   org.qt-project.QtWidgets      	0x0000000102363a30 QComboBoxPrivate::setCurrentIndex(QModelIndex const&) + 736
6   org.qt-project.QtWidgets      	0x0000000102362b2b QComboBoxPrivate::_q_rowsRemoved(QModelIndex const&, int, int) + 379
7   org.qt-project.QtWidgets      	0x00000001023692b2 QComboBox::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) + 1106
8   org.qt-project.QtCore         	0x0000000100e9e7ab QMetaObject::activate(QObject*, int, int, void**) + 2987
9   org.qt-project.QtCore         	0x0000000100f19dae QAbstractItemModel::rowsRemoved(QModelIndex const&, int, int, QAbstractItemModel::QPrivateSignal) + 78
10  org.qt-project.QtCore         	0x0000000100e1f9d0 QAbstractItemModel::endRemoveRows() + 80
11  org.qt-project.QtGui          	0x00000001019de07e QStandardItem::removeRows(int, int) + 510
12  org.qt-project.QtGui          	0x00000001019e1408 QStandardItemModel::removeRows(int, int, QModelIndex const&) + 88
13  org.qgroundcontrol.qgroundcontrol	0x0000000100154820 DebugConsole::removeLink(LinkInterface*) + 352
14  org.qt-project.QtCore         	0x0000000100e9e3c2 QMetaObject::activate(QObject*, int, int, void**) + 1986
15  org.qgroundcontrol.qgroundcontrol	0x00000001002ccf80 LinkManager::linkDisconnected(LinkInterface*) + 64
16  org.qt-project.QtCore         	0x0000000100e97393 QObject::event(QEvent*) + 755
17  org.qt-project.QtWidgets      	0x000000010226532b QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
18  org.qt-project.QtWidgets      	0x0000000102268648 QApplication::notify(QObject*, QEvent*) + 8136
19  org.qt-project.QtCore         	0x0000000100e6ca5b QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 971
20  libqcocoa.dylib               	0x0000000105304b8e QCocoaEventDispatcherPrivate::processPostedEvents() + 190
21  libqcocoa.dylib               	0x0000000105305411 QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void*) + 33
22  com.apple.CoreFoundation      	0x00007fff9127c7c1 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
23  com.apple.CoreFoundation      	0x00007fff9126e94d __CFRunLoopDoSources0 + 269
24  com.apple.CoreFoundation      	0x00007fff9126df7f __CFRunLoopRun + 927
25  com.apple.CoreFoundation      	0x00007fff9126d998 CFRunLoopRunSpecific + 296
26  com.apple.HIToolbox           	0x00007fff898707df RunCurrentEventLoopInMode + 235
27  com.apple.HIToolbox           	0x00007fff8987045e ReceiveNextEventCommon + 179
28  com.apple.HIToolbox           	0x00007fff8987039b _BlockUntilNextEventMatchingListInModeWithFilter + 71
29  com.apple.AppKit              	0x00007fff935d5a6d _DPSNextEvent + 964
30  com.apple.AppKit              	0x00007fff935d5220 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 194
31  com.apple.AppKit              	0x00007fff938dc197 -[NSApplication _realDoModalLoop:peek:] + 666
32  com.apple.AppKit              	0x00007fff93915af4 -[NSApplication runModalSession:] + 71
33  libqcocoa.dylib               	0x000000010530423a QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 1994
34  org.qt-project.QtCore         	0x0000000100e6937d QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381
35  org.qt-project.QtWidgets      	0x000000010244a192 QDialog::exec() + 514
36  org.qgroundcontrol.qgroundcontrol	0x000000010010f2c1 QGCFileDialog::getSaveFileName(QWidget*, QString const&, QString const&, QString const&, QString const&, bool, QFlags<QFileDialog::Option>) + 321
37  org.qgroundcontrol.qgroundcontrol	0x000000010010a287 QGCApplication::saveTempFlightDataLogOnMainThread(QString) + 199
38  org.qgroundcontrol.qgroundcontrol	0x00000001000d8083 void QtPrivate::FunctionPointer<void (QGCApplication::*)(QString)>::call<QtPrivate::List<QString, void>, void>(void (QGCApplication::*)(QString), QGCApplication*, void**) + 67
39  org.qt-project.QtCore         	0x0000000100e9e3c2 QMetaObject::activate(QObject*, int, int, void**) + 1986
[... cut ...]
  rax: 0x6b6f6f4200000000  rbx: 0x00000001108d2730  rcx: 0x000000000000002e  rdx: 0x00000001003bab47
  rdi: 0x00007fff5fbfb198  rsi: 0x0000000112c65b70  rbp: 0x00007fff5fbfb180  rsp: 0x00007fff5fbfb180
   r8: 0x00000001003bab48   r9: 0x00000000fe000000  r10: 0x00000001003bab48  r11: 0x0000000110f00000
  r12: 0x000000010d3cafe0  r13: 0xe1007eb46034c9a3  r14: 0x000000010d3cd8c0  r15: 0x00007fff5fbfb1b0
  rip: 0x00000001000f69d2  rfl: 0x0000000000010202  cr2: 0x00000001108275d8
  
Logical CPU:     0
Error Code:      0x00000000
Trap Number:     13
```

LorenzMeier, 2015-05-10:
Now I re-tried to download a different file, the Firmware side gives me this error for download: Nak received, error: no sessions available.

DonLakeFlyer, 2015-05-10:
> Nak received, error: no sessions available.

That makes sense. Since your last download didn't finish, firmware still has the session open for that one. It doesn't have any sort of timeout mechanism to close dropped connections.

LorenzMeier, 2015-05-10:
Let me know if I can contribute anything in addition - is there instrumentation / logging that I should be enabling?

DonLakeFlyer, 2015-05-10:
I'll have to look through the code to see what you can do to get more info. Need to think about it. The disconnect crash is more of the bugs from uas disappearing out from under things.

LorenzMeier, 2015-05-10:
I still don't get the need for deleting the UAS object instead of putting it into a disconnect state. Doing this properly would almost require to base all C++ controller code to derive from a base class which does "UAS gone" cleanup properly and consistently (or using reference counting, which is similar to putting the UAS into a disconnect state). I don't see the architectural drawback of having an unconnected UAS still instantiated, which refuses to communicate, but which avoids runtime failures.

Its not about correctness, its about defensive programming and a system architecture which either enforces the level of rigor the object deletion requires (base class which disconnects all signals / slots and provides a mechanism to stop all future accesses to the object), reference counting or keeping the object around.
What are we winning by deleting the object right now?

DonLakeFlyer, 2015-05-11:
You would have to change every API into the uas to do nothing, or maybe error out when it is disconnected. There are a massive set of apis there that currently have no idea what to do if the connection is gone. So you would need to fix up all those apis. There is currently no such thing as a uas which refuses to communicate at this point. 

If you do that and then don't change the ui. You end up with a pile of ui that looks like it should do something but when you click things it just doesn't respond. That's pretty horrible from a usability standpoint. So then you would still need change all the ui code such that it understands the uas that it has a reference to is disconnected and adjust the ui to a reasonable state. Otherwise users start to click things thinking something is going to happen and they just do nothing.

Hence the move to delete disconnected uas instead of trying to cauterize both the apis and the old ui somehow. Instead just remove ui (or optionally show a different ui) when there is no active uas available.

The new ViewWidget qml control and ViewWidgetController already handles all this uas switching for multiple vehicle cases and deleting correctly. These are the base classes you talk about above as being needed and they are already implemented. You just use ViewWidget as a base for your new Qml widget and if you need a controller you use ViewWidgetController as the base class for you C++ controller. You have to do next to no work yourself to make this all work correctly. 

There is no such base class support yet for top level views. That may come in the future if needed. Top level views tend to need to handle an active uas or no uas in a more special way. So it may not make sense for a base class. For example mission planning needs to know whether it has an active uas or not in order to work online or offline for waypoint editing. When disconnected it doesn't just do nothing, it works in completely different mode.

Only the old C++ widget code has problems with the concept of uas going away. In reality it is next to the same as the handling for the active uas changing. Almost all existing C++ widgets try to handle the NULL active uas case correctly. But in the old widgets, both switching active uas and having the active uas go to NULL was/is very buggy.  That said I've cleaned up most of the old widgets as well to fix the problems. Looks like there are still some holes there. If I remember correctly for example the DebugConsole code in which you are seeing the crash couldn't handle switching active uas from one to another (non-null) either. It would crash then as well. I fixed that, but looks like I still missed something. All of this is an interesting reason to toss the old widgets as fast as possible because in general they are very fragile to all sorts of things, not only this.

This will all go away as more things move to Qml widgets and use the ViewWidget framework. The Parameter Editor widget is the first new Qml dock widget. If you look at it's Qml code and controller you'll see that there is no code which handles all the switching and deleting uas thing. The code knows nothing about that. The ViewWidget base classes do all the work for it, keeping the main Qml/Controller simple and clean.

It's a bit painful to still be in the middle of a massive transition of the codebase. But in the end this will all end up being way easier to deal with.

DonLakeFlyer, 2015-05-11:
I can get very long-winded! 

To simplify even further: There are something like 6 or 7 old dock widgets that may or may not handle this correctly. That means 6 or 7 places that need to be changed to handle switching correctly. Majority of which have all been fixed. All in old code which is going to be tossed. Versus something like ~100 entry points into UASInterface apis that would need to be modified to handle a missing connection. All in code which is going to continue to live. The above approach is actually a less risky approach from a standpoint of missing things. You trade off possible dereference through a deleted UASInterface* in a few places, to a very large of set places that could dereference through a deleted or disconnected LinkInterface*.

DonLakeFlyer, 2015-05-11:
> It's a 100% repro on a messages file with 29992 bytes size.

It's a boundary case on filling the last packet fully and not correctly detecting eof. When I restructured all this code I disabled to unit test to fix in a separate pull. The UT of course tests all these boundary cases. Stupid move on my part.

LorenzMeier, 2015-05-12:
On-hardware testing confirms the fix, thanks!

closed 2015-05-12 by LorenzMeier