mavlink/qgroundcontrol #1551: (Mildly) Uncruft codebase before next stable / beta release

LorenzMeier, 2015-05-09:
I would propose to remove a few widgets which are fundamentally broken and continue to confuse users:

  * Info View: It never was properly finished and overlaying the same information on the map like @dogmaphobic did in the last iterations is much more useful. Classes to remove:   * https://github.com/mavlink/qgroundcontrol/blob/a5dc84680f871bcff37959819898930aa50e5c6f/src/ui/QGCTabbedInfoView.h
  * https://github.com/mavlink/qgroundcontrol/blob/a5dc84680f871bcff37959819898930aa50e5c6f/src/ui/uas/UASQuickView.h
  * https://github.com/mavlink/qgroundcontrol/blob/a4af4d4dd612d77dbb62d9f7c9af17388f946d80/src/ui/uas/UASMessageView.h

I have more of those, but this is the most visible one. Along with it we probably should hide the "Communication Console" widget. This one needs to be refactored to become a shell frontend using QML.

LorenzMeier, 2015-05-09:
@DonLakeFlyer In case you want to go ahead please feel free 8)

DonLakeFlyer, 2015-05-10:
Should we toss both Communication Console and Terminal until we get a real "Terminal" widget? I don't think either of these are working correctly .

LorenzMeier, 2015-05-10:
ok with me, yes.

Susurrus, 2015-05-10:
The only nice thing about the Communication Console is that it shows the up and down stream bandwidth. Besides that, I've never either of them.

severinleuenberger, 2015-05-10:
I have the communication console in my planning window per default. For me, the up/downstream are valuable and also to look up for error messages. Its easier than the drop down box in the list above. 
And no - there are not soo many bugs in it. I find it quite useful.

DonLakeFlyer, 2015-05-10:
We should move the up/downstream stuff to somewhere else.

DonLakeFlyer, 2015-05-10:
> And no - there are not soo many bugs in it. I find it quite useful.

You can communicate with nsh shell using it? That is what is is really meant for.

severinleuenberger, 2015-05-10:
...nope. nsh commands doesnt work. Didnt know, that this should be possible. Would be great of course...

DonLakeFlyer, 2015-05-25:
Done

closed 2015-05-25 by DonLakeFlyer