mavlink/qgroundcontrol #1554: Parameter meta data does not get updated locally after flashing master

LorenzMeier, 2015-05-11:
The parameter.xml file from the build system has the multicopter position control params (MPC_*) all in the right group, but QGC puts them into the default group:

[image]

DonLakeFlyer, 2015-05-11:
The MC_* parameters are duplicated in the xml file. The reason for this is that there are two separate C files in firmware which have the same set of MC_* params defined in them (modules/mc_att_control and modules/mc_att_control_mulitplatform). When QGC finds a duplicate of parameter meta data it assumes it can't trust any of the meta data for this param and tosses it. Hence it falls back to the default group.

If you can change the firmware such that there aren't two different C files with the same params in it this problem would go away.

LorenzMeier, 2015-05-13:
I just fixed this in Firmware. Could you propose an approach to auto-update the default values in QGC? Is that currently a manual commit?

kd0aij, 2015-05-13:
I suggest the "default group" be augmented with "broken" to reduce user expectations :)

DonLakeFlyer, 2015-05-13:
> Is that currently a manual commit?

Yes it is. It would require some sort of background download process to pull somewhere from the internet. Reason being you don't want to have to wait for it before you can connect to a vehicle. Also changing the meta data out from under a connected vehicle is not currently possible. Meta data is only loaded once, the first time a vehicle connects. So far example as it works now, if you have connected, disconnected and the flash new firmware you won't get new meta data till next QGC boot. This is all fixable and changeable but not super easy.

LorenzMeier, 2015-06-06:
Fixed on the FW side.

closed 2015-06-06 by LorenzMeier