mavlink/qgroundcontrol #1555: Param saved 5x

tumbili, 2015-05-11:


tumbili, 2015-05-11:
On initial connection via telemetry (using latest master on a quadrocopter) the "param saved" message appears about 5 times.

DonLakeFlyer, 2015-05-12:
The initial parameter download over telemetry isn't working quite correctly. It ends up re-requesting missed parameters too many times. In the end everything will be fine, since it will all settle down. But it can slow down connection quite a bit.

DonLakeFlyer, 2015-05-12:
You can still get some extra "settings autosaved" messages. But they should be more infrequent. The way QGC param loader is implemented it doesn't distinguish between parameters missing from initial load complete and waiting for a write param to come back. Because of that during initial param load it can end up generating extra param writes to storage. This greatly simplifies the QGC code. Don't want to change that since we had a number of problems with previous code trying to get to fancy with handling various states.

closed 2015-05-13 by tumbili