mavlink/qgroundcontrol #1556: Enable airspeed calibration for VTOL

tumbili, 2015-05-11:
Not entirely sure if this is a QGroundControl or a Firmware problem.

DonLakeFlyer, 2015-05-11:
Airspeed only shows up if vehicle is reported as MAV_TYPE_FIXED_WING. QGC doesn't understand VTOL as a fixed wing variant at this point. Needs to be fixed.

@LorenzMeier Seems like we need a better way to determine if airspeed cal should show up. Chasing changing MAV_TYPE's seems like a bad idea. Better to have some way to determine if an airspeed sensor is present.

Susurrus, 2015-05-11:
Why not add it as a sensor to the `common.xml:MAV_SYS_STATUS_SENSOR` enum? Then you can just switch on whether that sensor is reported. It'd also be nice to make it generic, so it could also indicate if there's a waterspeed sensor for a boat. The `ABSOLUTE_PRESSURE` and `DIFFERENTIAL_PRESSURE` might also be useful here, but I don't know.

LorenzMeier, 2015-05-12:
@Susurrus You have a really good point there. I need to think about how to do that properly.

DonLakeFlyer, 2015-05-12:
For now I'll trigger off of the VTOL MAV_TYPE's. But still need a better way.

DonLakeFlyer, 2015-05-25:
Done

closed 2015-05-25 by DonLakeFlyer