mavlink/qgroundcontrol #1559: Add option to calculate board rotation offset automatically

tumbili, 2015-05-12:
Users are finding it cumbersome to adjust the board rotation offset parameters manually(for fine tuning the horizon), especially because the values are in degrees and the euler angles coming from the autopilot are in radians.
It would be nice if the user could just press a button indicating that the plane is level now. QGroundControl could read the momentary attitude and calculate the rotation parameters accordingly.

tumbili, 2015-05-12:
I'm not 100% sure but I guess this is the reason why you are trying to get you plane level with APM before you power the autopilot.

severinleuenberger, 2015-05-12:
With APM it is really a headache, that you MUST level the plane before power up. I know several people (incuding myself) who nearly crashed a plane by power up on uneven ground and calibrate on wrong settings. 
When you are implementing this, then please just as an initial setup option. 

And to give the user a feedback, it would be helpful, to present on the GUI what settings have been saved (e.g. 2° positive pitch, 0.2° roll left.

tumbili, 2015-05-12:
@severinleuenberger Thanks for the feeback!

severinleuenberger, 2015-05-12:
...maybe add a warning, when roll-offset is above 5° (both sides) and pitch offset above 7°
Some airframe require higher offsets (the "Big Turtle" needs 9° pitch up offset), therefore it should just be a warning. But it could help detecting wrong settings. I had accidentally put the 9° on roll instead of pitch. The plane was still flyable, but had on automission constant offsets on the flypath ;-)

DonLakeFlyer, 2015-05-12:
I don't know anything about these settings. So you're going to need to provide me with full details on what parameters need to be set and how.

tumbili, 2015-05-12:
@DonLakeFlyer Yes, will do.

DonLakeFlyer, 2015-05-25:
Done

closed 2015-05-25 by DonLakeFlyer