mavlink/qgroundcontrol #1560: QGroundControl crashing issue

tumbili, 2015-05-12:
People from the dev call have been reporting that the main issue of QGroundControl is the fact that it tends to often crash, especially when disconnecting the autopilot. I think this problem is observed on Ubuntu 14.04 and Windows.

LorenzMeier, 2015-05-12:
I can second that. During normal use with the autopilot I quit QGC 80% of the time by crashing and only 20% are normal shutdowns. While this doesn't prevent daily use, I noticed that a crash is a remarkable event for the user (in particular if its the last thing he sees after the use). People I talk to report the app not being "ready" despite that it works great while they use it, simply because they feel any sign of instability is a no-no.

severinleuenberger, 2015-05-12:
I Recognized some crashes on Windows 7 (64bit) and Windows 8 (64 bit) on QGC downloaded 8 days ago.

DonLakeFlyer, 2015-05-12:
I'm going to need to details as to exactly when it crashes to try to fix anything. Crash on disconnect is likely the DebugConsole crash in Issue #1548. I'll fix that. Other than disconnect any more details on the other crashes?

DonLakeFlyer, 2015-05-12:
Sorry that was Issue #1550

DonLakeFlyer, 2015-05-12:
Can I toss "Communications Console", DebugConsole widget? From feedback it does two things right now:
1) Shows up/down link rates
2) Shows mavlink messages

It's doesn't really allow you to communication with nsh which is it's main purpose. I propose:
1) Toss "Communication Console" widget and Terminal view (neither really work)
2) Convert InfoView to Qml and make it really work (both the mavlink values and the message display)
3) Write a new Terminal tool/view that works
4) Move up/down link rates to someplace else (open to suggestions)

I'd like to toss DebugConsole right away though instead of having to fix the crashes.

DonLakeFlyer, 2015-05-25:
I haven't gotten any more details on Windows crashes. So now that DebugConsole is gone I'm closing this.

closed 2015-05-25 by DonLakeFlyer