mavlink/qgroundcontrol #1574: QGroundControl does connect to the PX4 Bootloader

LorenzMeier, 2015-05-14:
@dogmaphobic Its possible to connect to the boot loader, which is an issue as the connection will go when the main app starts and it seems as if it would just not work. This means that an user connecting in the first five seconds will not get a successful connection.

closed 2015-05-14 by LorenzMeier

dogmaphobic, 2015-05-14:
I'm not sure I follow. Are you saying that QGC is connecting to the boot loader? @DonLakeFlyer added code specifically to detect the boot loader and wait for it to go away before attempting to connect. What are the steps to reproduce this and how do you know it is connected to the boot loader?

With that being said, I have noticed in the past few weeks that sometimes I cannot connect the first time I start the PX4, regardless how long I wait. That is, I start the PX4, launch QGC and wait a bit (well over 5 seconds) and when trying to connect, it *connects* but nothing comes from the serial port. Restarting QGC and trying again always work.

LorenzMeier, 2015-05-14:
Look at the commit for the fix. It definitely did list the boot loader and you were able to connect to it. Its fixed now.

dogmaphobic, 2015-05-14:
Got it. That may explain what I've been seeing all along.

DonLakeFlyer, 2015-05-14:
> It definitely did list the boot loader and you were able to connect to it.

The serial link connect code waits to get past the bootloader before it opens the port. So it's ok if you click Connect while the bootloader is waiting. What was working incorrectly?

    // If we are in the Pixhawk bootloader code wait for it to timeout
    if (_isBootloader()) {
        qCDebug(SerialLinkLog) << "Not connecting to a bootloader, waiting for 2nd chance";
        const unsigned retry_limit = 12;
        unsigned retries;
        for (retries = 0; retries < retry_limit; retries++) {
            if (!_isBootloader()) {
                QGC::SLEEP::msleep(500);
                break;
            }
            QGC::SLEEP::msleep(500);
        }
        // Check limit
        if (retries == retry_limit) {
            // bail out
            qWarning() << "Timeout waiting for something other than booloader";
            return false;
        }
    }

LorenzMeier, 2015-05-14:
It never seemed to manage to connect then (it was connected, but got no data). I would propose to keep the current state where the link shows up only once the unit is ready and to see wether that's acceptable for users. On the plus side it is consistent with the Leds of the unit.

DonLakeFlyer, 2015-05-14:
> I would propose to keep the current state where the link shows up only once the unit is ready and to see wether that's acceptable for users.

It's a good change. Without it there is a strange delay before it "really" connects.

> It never seemed to manage to connect then (it was connected, but got no data).

That's happens to me quite a bit if I leave my board connected via USB and I'm testing things which requires doing lots of connects/disconnects/connects/.. At some point it will connect to the port, but it seems like mavlink never starts up so QGC never gets a heartbeat and it just hangs there.

commit f7afcef8 (2015-05-14): LinkManager: Do not connect to PX4 boards in bootloader mode automatically. Fixes #1574