mavlink/qgroundcontrol #1576: Crash after upgrading to current PX4/Firmware master and hitting connect

LorenzMeier, 2015-05-14:
QGC had a connection before this happened, which was disconnected by pulling on the cable first and then clicking disconnect.

```
Process:               qgroundcontrol [10025]
Path:                  /Users/USER/*/qgroundcontrol.app/Contents/MacOS/qgroundcontrol
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               2.0.0 (227)
Code Type:             X86-64 (Native)
Parent Process:        Qt Creator [199]
Responsible:           Qt Creator [199]
User ID:               502

Date/Time:             2015-05-14 18:00:48.315 +0200
OS Version:            Mac OS X 10.10.2 (14C2055)
Report Version:        11
Anonymous UUID:        FFA63A60-490C-41EB-AFE8-E82F56014EB6

Sleep/Wake UUID:       F4E488C9-441D-48CE-8DFD-CAF25A14A783

Time Awake Since Boot: 16000 seconds
Time Since Wake:       740 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000020

VM Regions Near 0x20:
--> 
    __TEXT                 0000000100000000-0000000100801000 [ 8196K] r-x/rwx SM=COW  /Users/USER/*/qgroundcontrol.app/Contents/MacOS/qgroundcontrol

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtCore         	0x0000000100e8e5e1 QVariant::QVariant(QVariant const&) + 17
1   org.qgroundcontrol.qgroundcontrol	0x00000001002a18f2 Fact::value() const + 18
2   org.qgroundcontrol.qgroundcontrol	0x000000010027f15f PowerComponent::setupComplete() const + 111
3   org.qgroundcontrol.qgroundcontrol	0x00000001002f603d VehicleComponent::qt_metacall(QMetaObject::Call, int, void**) + 429
4   org.qgroundcontrol.qgroundcontrol	0x00000001002f37ec PX4Component::qt_metacall(QMetaObject::Call, int, void**) + 28
5   org.qt-project.QtQml          	0x0000000101d6a317 QV4::QObjectWrapper::getProperty(QObject*, QV4::ExecutionContext*, QQmlPropertyData*, bool) + 3303
6   org.qt-project.QtQml          	0x0000000101d6941c QV4::QObjectWrapper::getQmlProperty(QV4::ExecutionContext*, QQmlContextData*, QV4::String*, QV4::QObjectWrapper::RevisionMode, bool*, bool) + 636
7   org.qt-project.QtQml          	0x0000000101d6d266 QV4::QObjectWrapper::get(QV4::Managed*, QV4::String*, bool*) + 70
8   org.qt-project.QtQml          	0x0000000101d78f29 QV4::Runtime::getProperty(QV4::ExecutionContext*, QV4::ValueRef, QV4::String*) + 105
9   ???                           	0x0000000105991bc4 0 + 4388887492
10  org.qt-project.QtQml          	0x0000000101d2a190 QV4::SimpleScriptFunction::call(QV4::Managed*, QV4::CallData*) + 528
11  org.qt-project.QtQml          	0x0000000101e1045e QQmlJavaScriptExpression::evaluate(QQmlContextData*, QV4::ValueRef, QV4::CallData*, bool*) + 622
12  org.qt-project.QtQml          	0x0000000101e101da QQmlJavaScriptExpression::evaluate(QQmlContextData*, QV4::ValueRef, bool*) + 58
13  org.qt-project.QtQml          	0x0000000101e186c4 QQmlBinding::update(QFlags<QQmlPropertyPrivate::WriteFlag>) + 1316
14  org.qt-project.QtQml          	0x0000000101e27253 QQmlObjectCreator::finalize(QQmlInstantiationInterrupt&) + 339
15  org.qt-project.QtQml          	0x0000000101dafec0 QQmlIncubatorPrivate::incubate(QQmlInstantiationInterrupt&) + 1264
16  org.qt-project.QtQml          	0x0000000101daf883 QQmlEnginePrivate::incubate(QQmlIncubator&, QQmlContextData*) + 451
17  org.qt-project.QtQml          	0x0000000101e672b5 QQmlDelegateModelItem::incubateObject(QQmlComponent*, QQmlEngine*, QQmlContextData*, QQmlContextData*) + 181
18  org.qt-project.QtQml          	0x0000000101e6705a QQmlDelegateModelPrivate::object(QQmlListCompositor::Group, int, bool) + 1434
19  org.qt-project.QtQml          	0x0000000101e67324 QQmlDelegateModel::object(int, bool) + 84
20  org.qt-project.QtQuick        	0x0000000101370fe5 QQuickRepeaterPrivate::createItems() + 197
21  org.qt-project.QtQuick        	0x0000000101370e9e QQuickRepeater::componentComplete() + 78
22  org.qt-project.QtQml          	0x0000000101e2731b QQmlObjectCreator::finalize(QQmlInstantiationInterrupt&) + 539
23  org.qt-project.QtQml          	0x0000000101dace69 QQmlComponentPrivate::complete(QQmlEnginePrivate*, QQmlComponentPrivate::ConstructionState*) + 73
24  org.qt-project.QtQml          	0x0000000101dab039 QQmlComponentPrivate::completeCreate() + 41
25  org.qt-project.QtQml          	0x0000000101dac7c2 QQmlComponent::create(QQmlContext*) + 114
26  org.qt-project.QtQuickWidgets 	0x000000010220984e QQuickWidget::continueExecute() + 238
27  org.qt-project.QtQuickWidgets 	0x000000010220974c QQuickWidgetPrivate::execute() + 620
28  org.qgroundcontrol.qgroundcontrol	0x00000001001161a9 QGCQuickWidget::setSource(QUrl const&) + 25
29  org.qgroundcontrol.qgroundcontrol	0x000000010028f14f SetupView::summaryButtonClicked() + 95
30  org.qgroundcontrol.qgroundcontrol	0x000000010027460c AutoPilotPlugin::_pluginReadyChanged(bool) + 284
31  org.qt-project.QtCore         	0x0000000100e833c2 QMetaObject::activate(QObject*, int, int, void**) + 1986
32  org.qgroundcontrol.qgroundcontrol	0x00000001002f0f2d AutoPilotPlugin::pluginReadyChanged(bool) + 61
33  org.qgroundcontrol.qgroundcontrol	0x000000010027fb87 PX4AutoPilotPlugin::_pluginReadyPreChecks() + 295
34  org.qt-project.QtCore         	0x0000000100e833c2 QMetaObject::activate(QObject*, int, int, void**) + 1986
35  org.qgroundcontrol.qgroundcontrol	0x00000001002a9c49 ParameterLoader::_parameterUpdate(int, int, QString, int, int, int, QVariant) + 7849
36  org.qgroundcontrol.qgroundcontrol	0x00000001002fa5c4 ParameterLoader::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) + 468
37  org.qt-project.QtCore         	0x0000000100e837ab QMetaObject::activate(QObject*, int, int, void**) + 2987
[... 162 lines cut ...]
4   org.qt-project.QtCore         	0x0000000100c444d5 QThread::exec() + 117
5   org.qt-project.QtCore         	0x0000000100c47f63 QThreadPrivate::start(void*) + 339
6   libsystem_pthread.dylib       	0x00007fff90d8f268 _pthread_body + 131
7   libsystem_pthread.dylib       	0x00007fff90d8f1e5 _pthread_start + 176
8   libsystem_pthread.dylib       	0x00007fff90d8d41d thread_start + 13

Thread 17:: QQmlThread
0   libsystem_kernel.dylib        	0x00007fff937153fa __select + 10
1   org.qt-project.QtCore         	0x0000000100ea2d80 QEventDispatcherUNIXPrivate::doSelect(QFlags<QEventLoop::ProcessEventsFlag>, timespec*) + 672
2   org.qt-project.QtCore         	0x0000000100ea3e4a QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 234
3   org.qt-project.QtCore         	0x0000000100e4e37d QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381
4   org.qt-project.QtCore         	0x0000000100c444d5 QThread::exec() + 117
5   org.qt-project.QtQml          	0x0000000101e28f09 QQmlThreadPrivate::run() + 57
6   org.qt-project.QtCore         	0x0000000100c47f63 QThreadPrivate::start(void*) + 339
7   libsystem_pthread.dylib       	0x00007fff90d8f268 _pthread_body + 131
8   libsystem_pthread.dylib       	0x00007fff90d8f1e5 _pthread_start + 176
9   libsystem_pthread.dylib       	0x00007fff90d8d41d thread_start + 13

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x00007fff5fbfaac8  rcx: 0x0000000000000043  rdx: 0x000000000000000b
  rdi: 0x00007fff5fbfaac8  rsi: 0x0000000000000020  rbp: 0x00007fff5fbfaa60  rsp: 0x00007fff5fbfaa40
   r8: 0x00000001003b29d0   r9: 0x00000000fc000000  r10: 0x00000001003b29d0  r11: 0x000000011f400000
  r12: 0x000000011a2977b0  r13: 0xff004ea359486185  r14: 0x0000000000000020  r15: 0x00007fff5fbfacf3
  rip: 0x0000000100e8e5e1  rfl: 0x0000000000010202  cr2: 0x0000000000000020
```

DonLakeFlyer, 2015-05-14:
That is a crash from a missing Fact. One of these is missing:
    return _autopilot->getParameterFact("BAT_V_CHARGED")->value().toFloat() != 0.0f &&
        _autopilot->getParameterFact("BAT_V_EMPTY")->value().toFloat() != 0.0f &&
        _autopilot->getParameterFact("BAT_N_CELLS")->value().toInt() != 0;

PowerComponent isn't switched to FactPanel yet. So we’re kind of in a middle state with respect to missing parameters.

LorenzMeier, 2015-05-14:
I must have hit a perfect timing on this one. This is only possible if the system hasn't fully booted yet, but mavlink is already up and running.
I need to go back and work out a strategy for that.

DonLakeFlyer, 2015-05-14:
Need to delay param request list response till after fully booted.

LorenzMeier, 2015-05-14:
Yes, but what does fully booted mean? Obviously a "mavlink fini" shell call would do. Just trying to work out if that's the best I can do.

DonLakeFlyer, 2015-05-14:
No comment on optional parameters :)

LorenzMeier, 2015-05-14:
Yeah, but get out flying with the thing and wait for the first param transmit with the current ~500 params - then you'll understand :)

DonLakeFlyer, 2015-05-14:
You could rely on the meta data in QGC, always run the full param list, but only send params which are non-default. Then QGC always shows full 500 list. Not a major change on the QGC side. Changes mavlink spec though. That would be screaming fast.

DonLakeFlyer, 2015-05-14:
Add a new flag to param request list or something for this mode.

LorenzMeier, 2015-05-14:
It's what most people use, so there is limited use for a mode. Only GCS developers have USB connected all the time 8).

DonLakeFlyer, 2015-05-14:
Make new param_request_list_packed and new param_value_packed mavlink messages. You can pack 11 parameters into a single param_value_packed message. Right now there are 300+ optional params which come down out of 500+ total params? Using packed param_value you'd get still get 5x+ speedup over optional params. Sending the list across one param per packet wastes a ton of bandwidth.

dogmaphobic, 2015-05-14:
No, make a new do_what_i_am_thinking message. The message packet length is 0 bytes and returns no value... :)

LorenzMeier, 2015-05-14:
Fixed in Firmware with a timeout. Closing it for now.

closed 2015-05-14 by LorenzMeier

DonLakeFlyer, 2015-05-15:
> Make new param_request_list_packed and new param_value_packed mavlink messages.

And of course this won't work now that I think about it.