mavlink/qgroundcontrol #1578: Store MAVLink logs automatically

AndreasAntener, 2015-05-15:
The MAVLink log on the GCS side is usually not so important, so I think most users just discard the save dialog on disconnect/quit. But sometimes you only notice later on that it would be great to have a look at it again (especially the transmitted messages).

Proposal and feature request:
- remove the save dialog
- write mavlink logs automatically to a folder in user home with a timestamp

This ensures we always have than in case we need it, and we won't get annoyed by the save dialog anymore :) (also there will be no need anymore to ask the user where to store "unsaved" data after a QGC crash)

DonLakeFlyer, 2015-05-25:
Good idea. I'll change this.

LorenzMeier, 2015-07-22:
@DonLakeFlyer This is still pretty annoying (sorry to be that bold about it) a lot of clicking, and I never save the flight logs now because I'm used to quickly click cancel. I think this would greatly add convenience.

AndreasAntener, 2015-08-03:
Looks like part of this is implemented? :) I see temporary files. Are they copied somewhere with a timestamp?

dagar, 2015-08-03:
In general would it be possible to stop cluttering the home directory?
Everything could be saved to ~/.config/QGroundControl.org

DonLakeFlyer, 2015-08-03:
Nothing has changed here recently. Except that logs except that you are only prompted to save if you armed.

DonLakeFlyer, 2015-08-03:
> In general would it be possible to stop cluttering the home directory?

Cluttering the home directory with what?

AndreasAntener, 2015-08-03:
> Nothing has changed here recently. Except that logs except that you are only prompted to save if you armed.

Ah, got excited to soon then ;)

dagar, 2015-08-03:
Sorry to further hijack this issue, but I currently have
~/mapscache
~/QGroundControl
~/.config/QGroundControl.org

I personally think it would be better to use the hidden directory by default and only place things directly in the home directory when explicitly chosen by the user.

So for this issue the mavlink logs could be saved by timestamp in ~/.config/QGroundControl.org

DonLakeFlyer, 2015-08-03:
> ~mapscache
This should really move to the standard hidden file location: ~/.config ...

> ~/QGroundControl
This is created as a sub-directory of the Qt standard Documents location. It is configurable through preferences. These are for user visible files like logs, save parameters and so forth.

> ~/.config/QGroundControl.org
This is the standard Qt location for hidden files related to an application on Linux boxes. Qt is in control of this. In general you shouldn't have to mess with stuff in here.

DonLakeFlyer, 2015-08-03:
> ~mapscache
> This should really move to the standard hidden file location: ~/.config ...

@dagar Can you enter an issue on this one? We can fix this as we move to the new mapping code.

dagar, 2015-08-04:
> ~/QGroundControl
This is created as a sub-directory of the Qt standard Documents location. It is configurable through preferences. These are for user visible files like logs, save parameters and so forth.

Yes, but the directories were created and empty without ever saving parameters, etc. Hence my comment about clutter. Anyway, it's not a big deal.

Issue created for mapscache location.

LorenzMeier, 2015-10-12:
Kind of implemented based on arming.

closed 2015-10-12 by LorenzMeier