mavlink/qgroundcontrol #1580: Unable to open and install Qgroundcontrol

mxgx12, 2015-05-16:
Trying to get Qgroundcontrol installed on my Windows 8 machine so I can bind a satellite receiver to PIxhawk. I keep getting " Windows can't open this file" ... "What program do you want to use to open?". I've done my best to read through previous issues but haven't found an answer. I'm completely new to this GCS so please forgive any ignorance. Any insight into getting the program installed and running will be awesome!

dogmaphobic, 2015-05-16:
What exact file are you trying to open and where did you download it from?

mxgx12, 2015-05-16:
I'm trying open the windows download file (v2.3.0) from Qgroundcontrol.org
 
Date: Sat, 16 May 2015 01:48:29 -0700

dogmaphobic, 2015-05-16:
Hmmm... I downloaded it from here:

http://qgroundcontrol.s3.amazonaws.com/QGroundControl-Installer-Win32-Stable-V2.3.0.exe?dl=0

The installer run fine and installed it and when I launched it, it run just fine as well. Don't know what to say...

mxgx12, 2015-05-16:
When I download the file, it is has a file type of "0[1] File". IS this normal? I believe this is what's causing problems.

DonLakeFlyer, 2015-05-18:
> it is has a file type of "0[1] File"

Where are you seeing this?

DonLakeFlyer, 2015-05-19:
The links on the wiki page were out of whack for Windows. Try it again and it should work.

closed 2015-05-19 by DonLakeFlyer

mxgx12, 2015-05-19:
> https://github.com/mavlink/qgroundcontrol/issues/1580#issuecomment-103637388<p>The links on the wiki page were out of whack for Windows. Try it again and it should work.</p>
>    <link itemprop="url" href="https://github.com/mavlink/qgroundcontrol/issues/1580#issuecomment-103637388"></link>
>    <meta itemprop="name" content="View Issue"></meta>
>  <meta itemprop="description" content="View this Issue on GitHub"></meta>

Ok. Thanks. I got it to work. I appreciate your help. 

Matt McGowan