mavlink/qgroundcontrol #1584: Connection doesn't complete when using a radio.

dogmaphobic, 2015-05-18:
More than half the time, when I connect using a radio modem, the connection doesn't complete. The progress bar is almost at the end and it stays there stuck. The last lines in the console are:

```
Settings location "/Users/gus/.config/org.qgroundcontrol/QGroundControl.ini"
Saved files location "/Volumes/Data/Documents/QGroundControl"
""
"1 Input devices found:"
"	SpaceNavigator"
"	Number of Axes: 6"
"	Number of Buttons: 2"
Orphaned log file count 0
Temp log "/var/folders/k6/cxtnktrn32sf18mdxtm13msc0000gn/T/FlightData902399.mavlink"
TYPE CHANGED TO: 13
Add new UAS: 1
Unknown element in XML: "board"
/Users/gus/GitHub/qgroundcontrol/src/uas/UASWaypointManager.cc 1083 "Requesting waypoint list..."
Unknown message from system: 1 message: 87
```
[image]

LorenzMeier, 2015-05-18:
Could you re-flash with current stable Firmware and see if anything changes? I have 3/3 success rate via radio.

dogmaphobic, 2015-05-18:
I have to leave (that's why I didn't have time to dig further) but I will try later on. I'm using current master because of the battery consumption change (#1583).

LorenzMeier, 2015-05-18:
My screwup during a rework of the changed storage this morning. Fixed in: PX4/Firmware@c06ba04

closed 2015-05-18 by LorenzMeier