mavlink/qgroundcontrol #1586: alert-text instead of a symbol with exclamation mark

severinleuenberger, 2015-05-19:
I would prefer to see directly the mavlink error message instead of a red flag with exclamation mark and number of messages. 
On rollover, the function it can be as it is now. 
Default view should be: alwasys the last message. 
Something like this: 
[image]

LorenzMeier, 2015-05-19:
I don't think this will scale or be particularly readable. If we want to show them more prominently then we would need to do them as banner below the status bar or something similar.

LorenzMeier, 2015-06-06:
@dogmaphobic @DonLakeFlyer Most users miss the messages in the toolbar because a lot of icons are red there essentially just indicating "normal" things like no GPS lock. It would make sense to show a banner over the top of the current active Widget (so its still readable) with those messages for a few seconds.
Its a recurring pattern, and its extremely relevant for the overall user experience.

DonLakeFlyer, 2015-06-06:
Yesterday I flew with Tower on my Iris with the APM stack. It was pretty interesting. This whole QGC top bar thing needs to be reworked to be more user friendly and useful while flying. I've took a lot of notes for things to change.

LorenzMeier, 2015-10-12:
Implemented!

closed 2015-10-12 by LorenzMeier