mavlink/qgroundcontrol #1588: need Previous stable builds

birdiebertus, 2015-05-19:
hello please help i look for a Previous stable builds i want to instal it on windows 8 64 bit or on windows xp i try it whit version off website but do not work 
sorry for my bad engels
bertus from holland

DonLakeFlyer, 2015-05-19:
There are no previous stable builds. If you detail why the current version doesn't work maybe we can fix it.

birdiebertus, 2015-05-19:
Windows 8 whil not open the file

I don't no how to open the file

Gr Bertus

Verzonden van mijn HTC

----- Reply message -----

https://github.com/mavlink/qgroundcontrol/issues/1588#issuecomment-103569838

DonLakeFlyer, 2015-05-19:
I can't help you if you don't provide me with any details! Details means:
- What exactly does windows say when you try to open the file?
- What is the name of the file you are trying to open?
- How are you trying to open the file? Command line? Double-click somewhere?

birdiebertus, 2015-05-19:
Oké
THE file i try to open named qgroundcontrol-installer-win32-pixhawk

When i double click i get This
         
                                
 Sorry but my Engels is bad
Best bertus

DonLakeFlyer, 2015-05-19:
I assume there was supposed to be an image there, but there is no image?

birdiebertus, 2015-05-19:
Gr bertus

birdiebertus, 2015-05-19:
I dont no why you dont get THE image 

Gr bertus

birdiebertus, 2015-05-19:
Yello is it not posible to sent zip file whit instruktions i even want to pay for it 

Gr bertus

DonLakeFlyer, 2015-05-19:
You can't attach files from an email. You need to do it directly from github. Below the comment section you'll see a link which you click on to attach an image.

birdiebertus, 2015-05-19:
[image]
[image]
[image]

DonLakeFlyer, 2015-05-19:
Can you translate the message which is talking about SDL.DLL for me?

LorenzMeier, 2015-05-19:
My dutch is a bit rusty, but its saying it can't find SDL.DLL and proposes to reinstall the software. Are you packaging that one? We're using it for joystick support (and our desire for joystick support is growing since we need it to make full use of the SITL capabilities - I will fix up that joystick PR).

birdiebertus, 2015-05-19:
Program kan not be started it is misging SDL.DLL is missing from my computer system

DonLakeFlyer, 2015-05-19:
SDL is supposed

DonLakeFlyer, 2015-05-19:
AWS seems to have maybe changed something with how you link to downloads on Windows. The wiki links were out of whack. Can you try downloading a new version from the wiki? After it installs, if you still get the SDL.DLL error can you look in the c:\Program Files (x86)\qgroundcontrol directory and see if there is an SDL.DLL file in there?

birdiebertus, 2015-05-19:
You meen download from qgroundcontrol site download section?

DonLakeFlyer, 2015-05-19:
Yes

birdiebertus, 2015-05-19:
It runs many thanks !!

closed 2015-05-19 by DonLakeFlyer

birdiebertus, 2015-05-20:
Hello I have a problem I try to install qgroundcontrol on xp olso it installs but when I try to open the program  I get this c:/program files/qgroundcontrol.exe is no vailid win32-application
I hope you want to help me

Gr Bertus

Verzonden van mijn HTC

----- Reply message -----

Closed #1588.

---
https://github.com/mavlink/qgroundcontrol/issues/1588#event-308634095

DonLakeFlyer, 2015-05-20:
QGC does not run on windows xp.

birdiebertus, 2015-05-20:
Oke thanks

Gr Bertus

Verzonden van mijn HTC

----- Reply message -----

https://github.com/mavlink/qgroundcontrol/issues/1588#issuecomment-103903102