mavlink/qgroundcontrol #1590: Onboard Files Critical error when indexing

Austriker, 2015-05-20:
I am trying to download files with onboard files widget. 
Got critical error with when trying to index files on the SD card because of hidden file from spolight.

[image]

LorenzMeier, 2015-05-20:
Is that with a QGC daily build? If not, try upgrading first.

LorenzMeier, 2015-05-20:
@DonLakeFlyer should we add a list configure or a list-all command so we can hide hidden files by default but still grant access if needed?

Austriker, 2015-05-20:
It's a Daily Build from this morning.

DonLakeFlyer, 2015-05-24:
The problem is that the list command never completes right? If so, the failure handling for failed opendir or readdir is likely not working correctly on the firmware side. Causing the QGC ftp state machine to get confused.

As far as showing  hidden files, seems like nuttx can't deal with them?

DonLakeFlyer, 2015-05-24:
I see what's happening. It getting stuck in a loop. Not sure if FW or QGC at this point.

Austriker, 2015-05-25:
Tested works fine still one error message but you can download files.
[image]

DonLakeFlyer, 2015-05-25:
You'll still get the errors since those are real. But you'll be able to list the files.

closed 2015-05-25 by DonLakeFlyer