mavlink/qgroundcontrol #1592: flight summary after flight (for operators)

severinleuenberger, 2015-05-21:
When a flight is finished, it would be helpful to have a brief flight-summary as a popup-window on the screen. 

The trigger for the popup-window would be, when the aircraft is manually disarmed after flight. 

Very important to know for all who fly operations would be
- total distance flown
- total airtime 
- mAh used from battery
- Battery voltage when finished

nice to have 
- average Ampere consumption
- average flight 
- max distance from home
- number of "connections-lost" from groundstation
- max m/s airspeed during flight
- min m/s airspeed during flight

Anyone who could implement this?
Best, Severin

NiqueIt, 2015-05-21:
What is needed for any "official pilot log":
- date/time of takeoff and landing --> block time (from arm to disarm)
- location (geo-position)
- count of landings in between

And one thing that would be great, if you can download directly a kml-file.... 

Sorry for more requirements than any implementation on it. I can fly, test and put some requirements, but not coding :(

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier