mavlink/qgroundcontrol #1598: Add RC_MAP_ACRO to mode switch UI

LorenzMeier, 2015-05-23:
@DonLakeFlyer This one I completely forgot: We reconfigured the RC_MAP_ACRO switch into becoming part of the normal mode selection. If configured it will be evaluated once in manual mode.

Right now its a 2-position switch since we have no acro mode in fixed wing yet, but that would be trivial to add.

@tumbili FYI

Fixed wing:
 - off: manual
 - center: acro (NOT YET IMPLEMENTED IN FIRMWARE, but could be easily changed in sensors.cpp)
 - on: stabilized

Multicopter:
  - off: manual (= stabilized)
  - center: acro
  - on: acro

DonLakeFlyer, 2015-05-23:
So it needs to be on it's own separate channel?

DonLakeFlyer, 2015-05-23:
Confused as to how this switch can work with one RC_ACRO_TH defaulting to 0.5? That will only work with a two position switch. But it seems like you are describing this as a 3 position switch: off/center/on?

DonLakeFlyer, 2015-05-23:
Is there an update this this: https://pixhawk.org/users/system_modes?s[]=flight&s[]=mode, which would explain it?

DonLakeFlyer, 2015-05-24:
@LorenzMeier Any more info on this?

LorenzMeier, 2015-06-06:
Fixed with recent master, thanks!

closed 2015-06-06 by LorenzMeier