mavlink/qgroundcontrol #1608: connect Pixhawk over 3DR Radio fails on Linux with QGroundcontrol stable v2.4

nopeppermint, 2015-05-27:
does someone know a solution to this error?
QGroundcontrol stable 2.4 on linux 32bit, trying to connect to pixhawk with px4 flightstack stable
over 3dr radio, on pixhawk connected on Telem 1
[image]

DonLakeFlyer, 2015-05-27:
Did you check that your baud rates are the same on both radios?

dogmaphobic, 2015-05-27:
> Did you check that your baud rates are the same on both radios?

That would imply this is the first time these radios are being used. Is that the case? If it is indeed the first time these radios are being used, make sure they are both flashed with the latest firmware as well.

DonLakeFlyer, 2015-05-27:
Also keep in mind that QGC doesn’t support radio setup yet. You’ll need to use Mission Planner for that.

nopeppermint, 2015-05-27:
I tried with two different pairs of radios. Both didn't worked.
I used these Radios successfully before for a serial connection over the air @115200 baud.
=> check baud rate are the same on both radios=> Do I have to configure this manualy for each Radio (in Mission Planner)?
Isn't this done by QGroundcontrol (see my picture on the left /dev/ttyUSB1 57600 8 0 1) and by the pixhawk on the other side?

=> I will try the firmware update

DonLakeFlyer, 2015-05-27:
That is the baud rate that QGC is talking to the radio at. The baud rate that the radios talk to each other at is a different thing. That must be configured seperately.

nopeppermint, 2015-05-27:
ok, thanks, will see what the firmware update does ;-)

DonLakeFlyer, 2015-05-27:
FYI: I'll have 3DR Radio setup done in QGC in a few days.

dogmaphobic, 2015-05-27:
The reason I brought up the firmware issue is because the radios I got from 3DR worked out-of-the-box. Later I got a new set from eBay. They came with some old firmware and I only got them to work after flashing them with the current firmware.

nopeppermint, 2015-05-27:
Is there another way to flash them with the new firmware than over Mission Planner?
This seems to be the source code for the firmware https://github.com/Dronecode/SiK 
but there is a SiLabs USB debug adapter required. Isn't there a way to use some sort of USB bootloader on these devices (I think this is the way it's done inside Mission Planner, and maybe also in future QGC @DonLakeFlyer  ?)

nopeppermint, 2015-05-27:
Because I am not able to flash them with Mission Planner, I always got errors.

DonLakeFlyer, 2015-05-27:
I've never used anything other the Mission Planner. Make sure the radios are not connected to anything other than USB. I've found that strange things happen if they are connected to a board as well as USB.

nopeppermint, 2015-05-27:
I was searching for this https://github.com/mavlink/qgroundcontrol/issues/1338

dogmaphobic, 2015-05-27:
Ha! Yes, the Chinese (Air) radios don't come with USB. You need a (TTL level) serial/USB adapter. Something like this:

http://www.adafruit.com/products/954?gclid=CIbbyJO-4sUCFVSRHwodnTAAgA

dogmaphobic, 2015-05-27:
Oops, no, that one does not work. You need DTS/RTS and that one doesn't have it. This one does:

http://www.amazon.com/FT232RL-Serial-Adapter-Module-Arduino/dp/B00HSX3CXE

nopeppermint, 2015-05-27:
I am using original 3DR Radios, but they were shipped together with a Piksi DGNSS, so I think there may be another Firmware on them. So there is no need for your amazon and adafruit links ;-)

nopeppermint, 2015-05-27:
I was looking for #1338  because there was a link to http://www.pixhawk.org/peripherals/radio-modems/start#dr_usb_radio and that's what I am trying at the moment.

nopeppermint, 2015-05-27:
this is the actual github link https://github.com/Dronecode/SiK

DonLakeFlyer, 2015-05-27:
Let me know if you are running into problems there as well or what the ultimate solution is if you find one. I may need to take that into account when I implement radio firmware upgrade in QGC.

nopeppermint, 2015-05-27:
@DonLakeFlyer shure, I think it's the HW which comes preconfigured from piksi.. 
Will try with another one, directly from 3dr later this week.

nopeppermint, 2015-06-03:
@DonLakeFlyer 
just tested my new (now official 3DR Radios) with QGroundcontrol stable 2.4 on linux, they are working.
For the Sik Firmware update I tested the following setup:
```
git clone https://github.com/Dronecode/sik.git
cd Sik/Firmware
make install
tools/uploader.py --port /dev/ttyUSB0 dst/radio~hm_trp.ihx 
```
the terminal output then was:
```
uploading to port /dev/ttyUSB0
Connecting to /dev/ttyUSB0
Trying autosync

OK
ATI

ATI
SiK 1.9 on
AT&UPDATE
Sent update command
Got sync
board 4e  freq 43
erasing...
programing...
[##################################################] 42223/42223 (100%)

verifying...
[##################################################] 42223/42223 (100%)

done.
```
after this I reconnected the 3DR Radios with QGroundcontrol and they are still working, so the firmware update was successful.

One side note about the RSSI, with 3 digit it looks not so beautiful ;-)
[image]

LorenzMeier, 2015-06-03:
Thanks! Closing.

closed 2015-06-03 by LorenzMeier