mavlink/qgroundcontrol #1612: Parameter Editor shows "no vehicle connected" when connected via radio

kd0aij, 2015-05-28:
primary flight display and mavlink inspector show normal flow of mavlink data; artificial horizon is active and correct, live data plot works.

But Setup panel shows only the Firmware button on the left side.

latest master flashed using today's QGC daily build

dogmaphobic, 2015-05-28:
I just tested it and it's working fine. Note that it takes a while for QGC to download all parameters. Only after they are all downloaded will the Setup widgets be all available. On the other hand, a while back, the downloading was getting stuck towards the end. If that's the case, you need to update your firmware as well, though you mentioned you've done that.

The download progress bar indicates the current state of the download (and it is removed once the download completes).

[image]

kd0aij, 2015-05-28:
I'm getting no progress bar at all

LorenzMeier, 2015-05-29:
Are you sure communication works in both directions?

kd0aij, 2015-05-29:
There's definitely no data flowing from GCS to Pixhawk over the telemetry port. Moved the same cable and radio to another pixhawk with older PX4 firmware, and QGC downloads parameters normally. 
The older firmware is Firmware Version	0e0c500f03053da5ef2b8df7a1d71c789c7ea2db from late March,2015.

LorenzMeier, 2015-05-29:
Does stable work for you? Master works for me for most things, but has issues in HIL I'm aware of.

kd0aij, 2015-05-29:
same behavior with stable; will try on a different AP to see if it's a problem with the serial port

kd0aij, 2015-05-29:
OK looks like this must be a problem with the serial port on one of my X2s. No problem with with release_v1.0.0 installed on my FMUv1.

Closing

closed 2015-05-29 by kd0aij