mavlink/qgroundcontrol #1624: segfault on attempt to calibrate accelerometers with SENS_BOARD_ROT=28

kd0aij, 2015-05-31:
QGC 5/30 daily build, firmware branch release_v1.0.0

DonLakeFlyer, 2015-05-31:
Can you install the new daily build and then follow these instructions: http://www.qgroundcontrol.org/bugreporting <http://www.qgroundcontrol.org/bugreporting>. 

It doesn’t have a bug fix in it, but I’m trying to create a repeatable mechanism to debug user crashes on Windows.

kd0aij, 2015-05-31:
Ubuntu 14.10
QGC daily build 5/31
AUAV-X2 pixhawk derivative
PX4/Firmware branch release_v1.0.0
connect to pixhawk, go to Setup:Sensors, press Gyro, segfault with no error
messages

core dump https://copy.com/Yws6REf4rDkHSFON

kd0aij, 2015-06-02:
further testing shows that pressing any button on the Sensors page results in a segfault (after pressing OK) with the above setup, regardless of the value of the rotation parameter.

DonLakeFlyer, 2015-06-02:
Should be fixed in latest daily.

LorenzMeier, 2015-06-03:
@kd0aij Fixed in latest daily?

kd0aij, 2015-06-03:
OK on Ubuntu

closed 2015-06-03 by kd0aij