mavlink/qgroundcontrol #1639: Parameter categories

LorenzMeier, 2015-06-04:
This is a mixed Firmware / QGC thing: We should have categories like:

```C
/**
 * @category User
*  @category Expert
*  @category Developer
*/
```

So we can start to hide the non-user params in different sections. The category would not define what a user needs to setup, but what a user needs to change in the parameter editor (e.g. the RC* params would be in the Developer category, because this is something neither a user nor expert should be messing around manually with).

dagar, 2016-02-27:
I like this idea. Still interested?

DonLakeFlyer, 2016-02-28:
Not sure. If the majority of non power user parameter usage is coverd with config screens. Then the parameter editor is a power user feature. The question is whether for power use, this will just add noise/complexity to the interaction model.

dagar, 2016-02-28:
It might reduce the need for some configuration screens. For example the battery configuration screen is just the important params pulled out. If the user level params were heavily reduced to the basics would that have even been needed? Likewise for geofence.

If you decide to go this way I volunteer to take the first pass at adding all the metadata.

DonLakeFlyer, 2016-02-28:
> It might reduce the need for some configuration screens.

Yes, but I've had a different idea in the back of my head for that. It's sort of like this but more detailed. A way to let the firmware take more control over the parameter editing experience.

DonLakeFlyer, 2018-01-03:
https://github.com/PX4/Firmware/issues/8568

DonLakeFlyer, 2018-01-04:
Replaced by #5965

closed 2018-01-04 by DonLakeFlyer