mavlink/qgroundcontrol #1641: UI freezes during param download on Windows

LorenzMeier, 2015-06-04:
@DonLakeFlyer When connecting to Pixhawk via USB (not radio) the UI freezes (and Windows complains about a crashed app sometimes) during param download. This is the case for two different setups running Windows 7 and Windows 8.

It seems this doesn't happen when connected via Radio, which is weird enough.

DonLakeFlyer, 2015-06-06:
I've heard this multiple times but I can't repro it on my end. If you can get a crash following the crash reporting steps that would be great.

LorenzMeier, 2015-06-06:
It doesn't crash, it just freezes with the app freeze mouse indicator.

DonLakeFlyer, 2015-06-06:
That's worse since I can't debug from a stack trace. Can you break in and figure out where it is?

LorenzMeier, 2015-06-06:
I don't have Windows, I only have Windows users - we would need a Windows dev to do that.

DonLakeFlyer, 2015-06-06:
I need to expose the logging settings in the ui so you can turn things on/off. Then users could capture logs which would be better than nothing.

DonLakeFlyer, 2015-08-24:
I can repro this now. Still no idea as to how to debug though.

LorenzMeier, 2015-09-01:
@DonLakeFlyer Could our new Windows dev help there?

DonLakeFlyer, 2015-09-03:
Who is that?

closed 2015-10-25 by LorenzMeier