mavlink/qgroundcontrol #1642: QGroundcontrol stable 2.4 linux, always crash if I close UDP connection to pixhawk

nopeppermint, 2015-06-05:
QGroundcontrol stable 2.4 linux, always crash if I close UDP connection to pixhawk
Linux Mint 17 32bit, kernel 3.13.04, qgroundcontrol/qt54 installed from ppa

I got the following error (in the terminal) before closing:
```
MAVLinkProtocolLog: _linkStatusChanged "cb3d4e0" false
Memory access error (translated from german)
``` 
udp connection was to 127.0.0.1:14555

nopeppermint, 2015-06-05:
Two more last messages before crash:
```
MAVLinkProtocolLog: _linkStatusChanged "bed4330" false
QXcbWindow: Unhandled client message: "_GTK_LOAD_ICONTHEMES"
QXcbWindow: Unhandled client message: "_GTK_LOAD_ICONTHEMES"
QXcbWindow: Unhandled client message: "_GTK_LOAD_ICONTHEMES"
QXcbWindow: Unhandled client message: "_GTK_LOAD_ICONTHEMES"
Memory access error (translated from german)
```
```
MAVLinkProtocolLog: _linkStatusChanged "975ba10" false
Memory access error (translated from german)
```

LorenzMeier, 2015-06-05:
Could  you please try the daily? This should have been fixed there.

nopeppermint, 2015-06-05:
daily is only as x64 available :-(
I have x86, so I have to build on my own ?

nopeppermint, 2015-06-05:
build successful on my own, error is gone.
But the udp connection is plotted twice
[image]

LorenzMeier, 2015-06-05:
That's because you created the connection twice. You need to remove one of them from your settings. Thanks for reporting!

closed 2015-06-05 by LorenzMeier

nopeppermint, 2015-06-05:
>Thanks for reporting!

will v2.4 stable get updated or will it rest with errors?
(especially the ppa) 
I am asking just to prevent other user to get the same experience with a "stable" released version ;-)

DonLakeFlyer, 2015-06-05:
I'll be releasing a new Stable in a few days. Still no x86 though.

LorenzMeier, 2015-06-05:
Don, can we do the stable now? People continue to run into the known crashes.

DonLakeFlyer, 2015-06-05:
Probably by tonight. I'm headed out flying if you can believe that.