mavlink/qgroundcontrol #1643: Show waypoint AGL / AMSL altitude on the new style 2D map

LorenzMeier, 2015-06-05:
@dogmaphobic The old map suffers from zero situational awareness when it comes to altitude. It would be awesome if the new map would show the AGL altitude (based on the Google data?).

Nyaruko, 2015-06-05:
That would be awesome!  I am still using the old version. May I ask whether
the offline map is available now?

dogmaphobic, 2015-06-05:
I have to see how simple it is to get that (AGL altitude). It would be an http request for every position change to get the AGL for that specific coordinate. Let me look into it and see what I can come up with. 

As to offline map, not really. The map tiles *are* cached as you *load* them but there isn't yet a tool for you to determine an area and ask for it to pre-load the tiles. As is, the caching is all done by the Qt mapping service, which isn't exactly great. It's in my to-do list. Adding our own caching on top of Qt's so I have better control over all this. The upcoming Qt 5.5 has a ton of fixes in this area and I'm waiting for it before putting too much effort in fixing things on this side of the fence.

By the way, all this is based on the new mapping display shown on the main flight view. I'm not doing anything to the old, *Mission View*.

dogmaphobic, 2015-06-05:
Google has an API for this. In theory, you need a *key* (developer or commercial) and it has a volume limit:

https://developers.google.com/maps/documentation/elevation

Though if you try it with no key whatsoever, it seems to work (no idea for how long though):

https://maps.googleapis.com/maps/api/elevation/json?locations=37.803784,-122.462276

Nyaruko, 2015-06-06:
But this means you will have to access the internet for the height
information?

dogmaphobic, 2015-06-07:
> But this means you will have to access the internet for the height
> information?

The same as the map tiles. If this information is to be collected, it would also be cached. However, the more I look at it, the less practical I think it will be. For it to make any difference, it would have to have a minimum of a 1m precision. For that to happen, I would need to collect the altitude for every meter the vehicle moves. Real time updates for moving a straight line for 1km == 1,000 requests. Request to cache a 1km x 1km area == 1,000,000 requests. Yes, I could skip some and interpolate but the more that's done, the less accurate it becomes, to the point of being useless if you can't trust it. What we need is a radar altimeter...

dogmaphobic, 2015-06-07:
Something like this if you don't mind the 17kg (never mind the cost):

http://www.robotshop.com/en/sri500-precision-3d-laser-scanner.html

:)

LorenzMeier, 2015-06-07:
@dogmaphobic We only need the altitude for waypoint positions. So the requests would be much less.

DonLakeFlyer, 2015-06-07:
How does Mission Planner get heights do the terrain following thing?

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier