mavlink/qgroundcontrol #1644: XPlane HIL crashes on connect / disconnect

LorenzMeier, 2015-06-05:


LorenzMeier, 2015-06-05:
```
Process:               qgroundcontrol [596]
Path:                  /Users/USER/*/qgroundcontrol.app/Contents/MacOS/qgroundcontrol
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               2.0.0 (227)
Code Type:             X86-64 (Native)
Parent Process:        Qt Creator [187]
Responsible:           Qt Creator [187]
User ID:               502

Date/Time:             2015-06-05 22:08:47.891 +0200
OS Version:            Mac OS X 10.10.2 (14C2055)
Report Version:        11
Anonymous UUID:        FFA63A60-490C-41EB-AFE8-E82F56014EB6

Sleep/Wake UUID:       13C79A70-08B4-4BE8-9D43-FE13CD677F50

Time Awake Since Boot: 3800 seconds
Time Since Wake:       2500 seconds

Crashed Thread:        29  QGCXPlaneLink

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000001

VM Regions Near 0x1:
--> 
    __TEXT                 0000000100000000-0000000100802000 [ 8200K] r-x/rwx SM=COW  /Users/USER/*/qgroundcontrol.app/Contents/MacOS/qgroundcontrol

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff93c3a4de mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff93c3964f mach_msg + 55
2   com.apple.framework.IOKit     	0x00007fff9b37116c io_connect_method + 381
3   com.apple.framework.IOKit     	0x00007fff9b3149da IOConnectCallMethod + 349
4   com.apple.framework.IOKit     	0x00007fff9b314bde IOConnectCallStructMethod + 56
5   com.apple.IOAccelerator       	0x00007fff9b2e6b08 IOAccelContextSubmitDataBuffers + 245
6   libGPUSupportMercury.dylib    	0x00007fff8e52bf5a gpusSubmitDataBuffers + 142
7   com.apple.driver.AppleIntelBDWGraphicsGLDriver	0x0000123400003702 IntelCommandBuffer::getNew(GLDContextRec*) + 48
8   com.apple.driver.AppleIntelBDWGraphicsGLDriver	0x00001234000f2a11 intelSubmitCommands + 171
9   com.apple.driver.AppleIntelBDWGraphicsGLDriver	0x00001234000f288a gldPresentFramebufferData + 142
10  GLEngine                      	0x00007fff96ec1131 glSwap_Exec + 97
11  com.apple.opengl              	0x00007fff8e46ff9a CGLFlushDrawable + 66
12  com.apple.AppKit              	0x00007fff91ee5c14 -[NSOpenGLContext flushBuffer] + 27
13  org.qt-project.QtGui          	0x00000001016bcbd3 QOpenGLContext::swapBuffers(QSurface*) + 499
14  org.qt-project.QtGui          	0x00000001018fb863 QPlatformBackingStore::composeAndFlush(QWindow*, QRegion const&, QPoint const&, QPlatformTextureList*, QOpenGLContext*, bool) + 3523
15  org.qt-project.QtWidgets      	0x0000000102254cc9 QWidgetBackingStore::qt_flush(QWidget*, QRegion const&, QBackingStore*, QWidget*, QPoint const&, QPlatformTextureList*, QWidgetBackingStore*) + 585
16  org.qt-project.QtWidgets      	0x0000000102254ed1 QWidgetBackingStore::flush(QWidget*) + 81
17  org.qt-project.QtWidgets      	0x0000000102258d8a QWidgetBackingStore::doSync() + 4250
18  org.qt-project.QtWidgets      	0x00000001022590b3 QWidgetBackingStore::sync() + 499
19  org.qt-project.QtWidgets      	0x000000010227c0cc QWidgetPrivate::syncBackingStore() + 236
20  org.qt-project.QtWidgets      	0x000000010228bed4 QWidget::event(QEvent*) + 2836
21  org.qt-project.QtWidgets      	0x000000010238eb21 QMainWindow::event(QEvent*) + 929
22  org.qt-project.QtWidgets      	0x000000010224c32b QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
23  org.qt-project.QtWidgets      	0x000000010224f648 QApplication::notify(QObject*, QEvent*) + 8136
24  org.qt-project.QtCore         	0x0000000100e53a5b QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 971
25  libqcocoa.dylib               	0x000000010705db8e QCocoaEventDispatcherPrivate::processPostedEvents() + 190
26  libqcocoa.dylib               	0x000000010705e411 QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void*) + 33
27  com.apple.CoreFoundation      	0x00007fff8d65b7c1 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
28  com.apple.CoreFoundation      	0x00007fff8d64d94d __CFRunLoopDoSources0 + 269
29  com.apple.CoreFoundation      	0x00007fff8d64cf7f __CFRunLoopRun + 927
30  com.apple.CoreFoundation      	0x00007fff8d64c998 CFRunLoopRunSpecific + 296
31  com.apple.HIToolbox           	0x00007fff8bff27df RunCurrentEventLoopInMode + 235
32  com.apple.HIToolbox           	0x00007fff8bff255a ReceiveNextEventCommon + 431
33  com.apple.HIToolbox           	0x00007fff8bff239b _BlockUntilNextEventMatchingListInModeWithFilter + 71
34  com.apple.AppKit              	0x00007fff91985a6d _DPSNextEvent + 964
35  com.apple.AppKit              	0x00007fff91985220 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 194
36  com.apple.AppKit              	0x00007fff919790c3 -[NSApplication run] + 594
37  libqcocoa.dylib               	0x000000010705d2fd QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 2189
38  org.qt-project.QtCore         	0x0000000100e5037d QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381
39  org.qt-project.QtCore         	0x0000000100e5335a QCoreApplication::exec() + 346
[... 272 lines cut ...]
4   org.qt-project.QtCore         	0x0000000100c49f63 QThreadPrivate::start(void*) + 339
5   libsystem_pthread.dylib       	0x00007fff912b9268 _pthread_body + 131
6   libsystem_pthread.dylib       	0x00007fff912b91e5 _pthread_start + 176
7   libsystem_pthread.dylib       	0x00007fff912b741d thread_start + 13

Thread 29 Crashed:: QGCXPlaneLink
0   ???                           	0x0000000000000001 0 + 1
[... cut ...]
  rax: 0x0000000000000001  rbx: 0x0000000000000002  rcx: 0x0000000110184ad0  rdx: 0x00000001003ab317
  rdi: 0x000000010400cc00  rsi: 0x000000010400cc00  rbp: 0x0000000126180d70  rsp: 0x0000000126180a68
   r8: 0x00000001003ab360   r9: 0x0000000000000002  r10: 0x0000000000000030  r11: 0x000000012ff00000
  r12: 0x0000000126180e08  r13: 0x0000000006010024  r14: 0x00007fff7de71070  r15: 0x00000001003ab317
  rip: 0x0000000000000001  rfl: 0x0000000000010246  cr2: 0x0000000000000001
  
Logical CPU:     2
Error Code:      0x00000014
Trap Number:     14
```

DonLakeFlyer, 2015-06-06:
Can you detail out the exact repro steps?

LorenzMeier, 2015-06-06:
@kd0aij Thanks for reporting this, I think its fixed now on master after the last commit I pushed. Closing this, please let me know in case you have any further issues.

closed 2015-06-06 by LorenzMeier