mavlink/qgroundcontrol #1645: Flight mode config: If the acro switch is mapped it leads to an error message

LorenzMeier, 2015-06-05:
[image]

DonLakeFlyer, 2015-06-05:
That's a hell of an error message!

DonLakeFlyer, 2015-06-05:
Do you have time to tweak this? My repo is torn apart right now. Need to remove Acro switch from both attitude string arrays in FlightModesComponentController..c:

    // Make sure switches are not double-mapped
    
    QStringList attitudeParams, attitudeNames;
    
    attitudeParams << "RC_MAP_THROTTLE" << "RC_MAP_YAW" << "RC_MAP_PITCH" << "RC_MAP_ROLL" << "RC_MAP_FLAPS" << "RC_MAP_AUX1" << "RC_MAP_AUX2" << "RC_MAP_ACRO_SW";
    attitudeNames << "Throttle" << "Yaw" << "Pitch" << "Roll" << "Flaps" << "Aux1" << "Aux2" << "Acro";

Otherwise, will take me a little while to get to this.

DonLakeFlyer, 2015-06-05:
in FlightModesComponentController::_validateConfiguration(void)

LorenzMeier, 2015-06-05:
Ok, that hint is good enough. I will get onto this.

LorenzMeier, 2015-06-05:
Closed via a7a36c02a60ba3b1bab961a7a012b3794c5da778

closed 2015-06-05 by LorenzMeier