mavlink/qgroundcontrol #1667: Make current system state transparent to user on a use-case basis

LorenzMeier, 2015-06-16:
We do indicate the RAW GPS status and num of sats, but we do not translate this yet into a human-language concept of "are you sure you want to take off? You can only do manual modes". Of course we should *not* use text for that.

This came up as a pre-flight check concern, but its impossible to enforce this on the autopilot without knowing what the user intends to do.

DonLakeFlyer, 2015-08-24:
Is the concept of allowing to fly in Manual mode only really needed? Seems like that is just going to end up being confusing. Seems like a hard stop (disable arming) on bad GPS status would be better. You could add a circuit-breaker to override for power user non-gps testing.

DonLakeFlyer, 2015-12-03:
This is all different now

closed 2015-12-03 by DonLakeFlyer