mavlink/qgroundcontrol #1673: Parameter Sync Fails after Partial Completion, Doesn't Resume

jgoppert, 2015-06-20:
I've seen this issue on both ubuntu 14.04 and windows with the latest 2.5 release.

See attached documents, the failure rate is about 30%. We've tried switching radios as well.

[QGroundController Connectivity Test Results 6_18_2015.docx](https://github.com/mavlink/qgroundcontrol/files/164/QGroundController.Connectivity.Test.Results.6_18_2015.docx)

jgoppert, 2015-06-20:
Here is an excel document documenting some more testing.

[QGroundController Connectivity Test Results 6_18_2015.pdf](https://github.com/mavlink/qgroundcontrol/files/165/QGroundController.Connectivity.Test.Results.6_18_2015.pdf)

DonLakeFlyer, 2015-06-20:
Do you only see this over radio? No failure over direct Ian with same rev of firmware?

jgoppert, 2015-06-22:
We are also seeing this over direct usb connection. See the added extra tests at the end.

[QGroundController Connectivity Test Results 6_21_2015.docx](https://github.com/mavlink/qgroundcontrol/files/175/QGroundController.Connectivity.Test.Results.6_21_2015.docx)

DonLakeFlyer, 2015-06-22:
Can you run QGroundControl with the —full-logging command line option and then send me the console output on a failed parameter load.

jgoppert, 2015-06-22:
Passed first 3 times, failed on the 4th.

[image]

DonLakeFlyer, 2015-06-22:
The option starts with two dashes not one: “—full-logging”. Also when  you run it can you copy all the content from the command prompt window to a file. Not just a screen shot.

ChauvinLou, 2015-06-22:
Here is the log I captured using
C:\PROGRA~2\qgroundcontrol>qgroundcontrol.exe --full-logging 1> c:\Users\lchauvin\qgr.log 2>&1

Copy and paste from the file to the text stream below. (part 1) See next comment for part 2 as data stream is too long (exceeds 262144 characters)
QGroundController never finished the update and remains on config screen without updating u/i on left side with options normally available after successful parameter file load.

[I] at :0 - "LOADING LIGHT"
[I] at :0 - "Saved files location "C:/Users/lchauvin/Documents/QGroundControl""
[I] at :0 - """"
[I] at :0 - "Orphaned log file count 0"
[I] at :0 - "Create SerialLink  "\\.\COM7" 57600 0 0 8 1"
[I] at :0 - "portName:  "\\.\COM7""
[I] at :0 - "CONNECT CALLED"
[I] at :0 - "SerialLink: hardwareConnect to  "\\.\COM7""
[I] at :0 - "PortName    :  "COM7" Description :  "USB Serial Port""
[I] at :0 - "Manufacturer:  "FTDI""
[I] at :0 - "PortName    :  "COM3" Description :  "Intel(R) Active Management Technology - SOL""
[I] at :0 - "Manufacturer:  "Intel""
[I] at :0 - "PortName    :  "COM10" Description :  "M6e""
[I] at :0 - "Manufacturer:  "CDC Driver""
[I] at :0 - "Configuring port"
[I] at :0 - "CONNECTING LINK:  src\comm\SerialLink.cc 387 type: "" with settings "\\.\COM7" 57600 8 0 1"
[I] at :0 - "_linkStatusChanged "9fd5680" true"
[I] at :0 - "Temp log "C:/Users/lchauvin/AppData/Local/Temp/FlightData152761.mavlink""
[I] at :0 - "TYPE CHANGED TO: 14"
[I] at :0 - "addLink: "9fd5680""
[I] at :0 - "Add new UAS:  1"
[I] at :0 - "Request to refresh all parameters"
[I] at :0 - "Loading PX4 parameter fact meta data"
[I] at :0 - "Loading parameter meta data: "C:/Users/lchauvin/AppData/Roaming/QGroundControl.org/PX4ParameterFactMetaData.xml""
[I] at :0 - "Found group:  "Attitude EKF estimator""
[I] at :0 - "Found parameter name: "EKF_ATT_V3_Q0"  type: "FLOAT"  default: "1e-4""
[I] at :0 - "Found parameter name: "EKF_ATT_V3_Q1"  type: "FLOAT"  default: "0.08""
[I] at :0 - "Short description: "Body angular acceleration process noise""
[I] at :0 - "Found parameter name: "EKF_ATT_V3_Q2"  type: "FLOAT"  default: "0.009""
[I] at :0 - "Short description: "Acceleration process noise""
[I] at :0 - "Found parameter name: "EKF_ATT_V3_Q3"  type: "FLOAT"  default: "0.005""
[I] at :0 - "Short description: "Magnet field vector process noise""
[I] at :0 - "Found parameter name: "EKF_ATT_V4_R0"  type: "FLOAT"  default: "0.0008""
[I] at :0 - "Short description: "Gyro measurement noise""
[I] at :0 - "Found parameter name: "EKF_ATT_V4_R1"  type: "FLOAT"  default: "10000.0""
[I] at :0 - "Short description: "Accel measurement noise""
[I] at :0 - "Found parameter name: "EKF_ATT_V4_R2"  type: "FLOAT"  default: "100.0""
[I] at :0 - "Short description: "Mag measurement noise""
[I] at :0 - "Found parameter name: "EKF_ATT_ENABLED"  type: "INT32"  default: "1""
[I] at :0 - "Short description: "EKF attitude estimator enabled""
[I] at :0 - "Long description: "If enabled, it uses the older EKF filter.
However users can enable the new quaternion
based complimentary filter by setting EKF_ATT_ENABLED = 0.""
[I] at :0 - "Min: "0""
[I] at :0 - "Max: "1""
[I] at :0 - "Found parameter name: "ATT_J11"  type: "FLOAT"  default: "0.0018""
[I] at :0 - "Short description: "Moment of inertia matrix diagonal entry (1, 1)""
[I] at :0 - "Unit: "kg*m^2""
[I] at :0 - "Found parameter name: "ATT_J22"  type: "FLOAT"  default: "0.0018""
[I] at :0 - "Short description: "Moment of inertia matrix diagonal entry (2, 2)""
[I] at :0 - "Unit: "kg*m^2""
[I] at :0 - "Found parameter name: "ATT_J33"  type: "FLOAT"  default: "0.0037""
[I] at :0 - "Short description: "Moment of inertia matrix diagonal entry (3, 3)""
[I] at :0 - "Unit: "kg*m^2""
[I] at :0 - "Found parameter name: "ATT_J_EN"  type: "INT32"  default: "0""
[I] at :0 - "Short description: "Moment of inertia enabled in estimator""
[I] at :0 - "Long description: "If set to != 0 the moment of inertia will be used in the estimator""
[I] at :0 - "Min: "0""
[I] at :0 - "Max: "1""
[I] at :0 - "Found group:  "Attitude Q estimator""
[I] at :0 - "Found parameter name: "ATT_W_ACC"  type: "FLOAT"  default: "0.2""
[I] at :0 - "Short description: "Complimentary filter accelerometer weight""
[I] at :0 - "Min: "0""
[I] at :0 - "Max: "1""
[... 3076 lines cut ...]
[I] at :0 - "_waitingReadParamIndexMap: (4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223, 224, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 235, 236, 237, 238, 239, 240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255, 256, 257, 258, 259 [... cut ...] , 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182,

ChauvinLou, 2015-06-22:
[I] at :0 - "Received PARAM_VALUE "PWM_MAIN_REV1" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "PWM_MAIN_REV1" count: 334 index: 7 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: (8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223, 224, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 235, 236, 237, 238, 239, 240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255, 256, 257, 258, 259, 260, 261, 262, 263, 264, 265, 266, 267, 268, 269, 270, 271, 272, 273, 274, 275, 276, 277, 278, 279, 280, 281, 282, 283, 284, 285, 286, 287, 288, 289, 290, 291, 292, 293, 294, 295, 296, 297, 298, 299, 300, 301, 302, 303, 304, 305, 306, 307, 308, 309, 310, 311, 312, 313, 314, 315, 316, 317, 318, 319, 320, 321, 322, 323, 324, 325, 326, 327, 328, 329, 330, 331, 332, 333)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 326"
[I] at :0 - "waitingParamCount: 326"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "EKF_ATT_ENABLED" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "EKF_ATT_ENABLED" count: 334 index: 8 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: (9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223, 224, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 235, 236, 237, 238, 239, 240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255, 256, 257, 258, 259, 260, 261, 262, 263, 264, 265, 266, 267, 268, 269, 270, 271, 272, 273, 274, 275, 276, 277, 278, 279, 280, 281, 282, 283, 284, 285, 286, 287, 288, 289, 290, 291, 292, 293, 294, 295, 296, 297, 298, 299, 300, 301, 302, 303, 304, 305, 306, 307, 308, 309, 310, 311, 312, 313, 314, 315, 316, 317, 318, 319, 320, 321, 322, 323, 324, 325, 326, 327, 328, 329, 330, 331, 332, 333)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 325"
[I] at :0 - "waitingParamCount: 325"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "ATT_BIAS_MAX" QVariant(float, 0.05) 9"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "ATT_BIAS_MAX" count: 334 index: 9 mavType: 9 value: QVariant(float, 0.05) )"
[I] at :0 - "_waitingReadParamIndexMap: (10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223, 224, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 235, 236, 237, 238, 239, 240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255, 256, 257, 258, 259, 260, 261, 262, 263, 264, 265, 266, 267, 268, 269, 270, 271, 272, 273, 274, 275, 276, 277, 278, 279, 280, 281, 282, 283,  [... cut ...] 257, 258, 259, 260, 261, 262, 263, 264, 265, 266, 267, 268, 269, 270, 271, 272, 273, 274, 275, 276, 277, 278, 279, 280, 281, 282, 283, 284, 285, 286, 287, 288, 289, 290, 291, 292, 293, 294, 295, 296, 297, 298, 299, 300, 301, 302, 303, 304, 305, 306, 307, 308, 309, 310, 311, 312, 313, 314, 315, 316, 317, 318, 319, 320, 321, 322, 323, 324, 325, 326, 327, 328, 329, 330, 331, 332, 333)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParam

ChauvinLou, 2015-06-22:
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "GF_ALTMODE" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "GF_ALTMODE" count: 334 index: 121 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: (122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223, 224, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 235, 236, 237, 238, 239, 240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255, 256, 257, 258, 259, 260, 261, 262, 263, 264, 265, 266, 267, 268, 269, 270, 271, 272, 273, 274, 275, 276, 277, 278, 279, 280, 281, 282, 283, 284, 285, 286, 287, 288, 289, 290, 291, 292, 293, 294, 295, 296, 297, 298, 299, 300, 301, 302, 303, 304, 305, 306, 307, 308, 309, 310, 311, 312, 313, 314, 315, 316, 317, 318, 319, 320, 321, 322, 323, 324, 325, 326, 327, 328, 329, 330, 331, 332, 333)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 212"
[I] at :0 - "waitingParamCount: 212"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "GF_MODE" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "GF_MODE" count: 334 index: 122 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: (123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223, 224, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 235, 236, 237, 238, 239, 240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255, 256, 257, 258, 259, 260, 261, 262, 263, 264, 265, 266, 267, 268, 269, 270, 271, 272, 273, 274, 275, 276, 277, 278, 279, 280, 281, 282, 283, 284, 285, 286, 287, 288, 289, 290, 291, 292, 293, 294, 295, 296, 297, 298, 299, 300, 301, 302, 303, 304, 305, 306, 307, 308, 309, 310, 311, 312, 313, 314, 315, 316, 317, 318, 319, 320, 321, 322, 323, 324, 325, 326, 327, 328, 329, 330, 331, 332, 333)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 211"
[I] at :0 - "waitingParamCount: 211"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "NAV_DLL_CHSK" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "NAV_DLL_CHSK" count: 334 index: 123 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: (124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223, 224, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 235, 236, 237, 238, 239, 240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255, 256, 257, 258, 259, 260, 261, 262, 263, 264, 265, 266, 267, 268, 269, 270, 271, 272, 273, 274, 275, 276, 277, 278, 279, 280, 281, 282, 283, 284, 285, 286, 287, 288, 289, 290, 291, 292, 293, 294, 295, 296, 297, 298, 299, 300, 301, 302, 303, 304, 305, 306, 307, 308, 309, 310, 311, 312, 313, 314, 315, 316, 317, 318, 319, 320, 321, 322, 323, 324, 325, 326, 327, 328, 329, 330, 331, 332, 333)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 210"
[I] at :0 - "waitingParamCount: 210"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "NAV_DLL_N" QVariant(int, 2) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "NAV_DLL_N" count: 334 index: 124 mavType: 6 value: QVariant(int, 2) )"
[I] at :0 - "_waitingReadParamIndexMap: (125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223, 224, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 235, 236, 237, 238, 239, 240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255, 256, 257, 258, 259, 260, 261, 262, 263, 264, 265, 266, 267, 268, 269, 270, 271, 272,  [... cut ...]
[I] at :0 - "_waitingReadParamIndexMap: (128, 314, 315, 316, 317, 318, 319, 320, 321, 322, 323, 324, 325, 326, 327, 328, 329, 330, 331, 332, 333)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 21"
[I] at :0 - "waitingParamCount: 21"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "CAL_GYRO0_YSCALE" QVariant(float, 1) 9"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "CAL_GYRO0_YSCALE" count: 334 index: 314 mavType: 9 value: QVariant(float, 1) )"
[I] at :0 - "_waitingReadParamIndexMap: (128, 315, 316, 317, 318, 319,

DonLakeFlyer, 2015-06-23:
@LorenzMeier Not sure what is going on here. You can see a the end of the log, the firmware just won't send comp:50, index:128 parameter to QGC even though it keeps requesting it. Maybe some boundary case problem with the bit twiddling for the used parameter map thing in the firmware?

LorenzMeier, 2015-06-23:
@jgoppert: Please provide the output of:
```
nsh> param index_used 128
index 128: x   INAV_SONAR_ERR [128,296] : 0.5000
```

This will be a different param name. It should have the [x] marking in front of it. Also, please cherry-pick this commit into your tree and re-try again: c192398

Make sure to get us the QGroundControl MAVLINK / TEXT console output. It should now complain on the UI level in the toolbar about param transmission fails, which will make it easier to debug.

ChauvinLou, 2015-06-23:
Not really clear what you are asking.
Using Terraterm on Windows and at nsh> param index_used 128
I get 

param: expect a command, ... see attachment for screen shot.

[image]

LorenzMeier, 2015-06-23:
what is the output for the same command running on index 127 and 129?

LorenzMeier, 2015-06-23:
If none of these work (e.g. also not index 1) then you're using a months-outdated version of the firmware and should upgrade, which will make your problem go away. Let us know in any case please.

ChauvinLou, 2015-06-24:
Loaded both
px4fmu-v2-default.px4 and
firmware-15-06-23-c4d54.px4 (built by James Goppert)

Get the following from both
nsh> param
param: expected a command, try 'load', 'import', 'show', 'set', 'compare', 'select' or 'save'

Also both still will fail parameter file load attempts on fourth or fifth try.

Bottom of output from QGroundControl --full-logging

---------- ....
[I] at :0 - "Received PARAM_VALUE "CBRK_RATE_CTRL" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "CBRK_RATE_CTRL" count: 336 index: 333 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: (172, 334, 335)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 3"
[I] at :0 - "waitingParamCount: 3"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "CBRK_SUPPLY_CHK" QVariant(int, 894281) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "CBRK_SUPPLY_CHK" count: 336 index: 334 mavType: 6 value: QVariant(int, 894281) )"
[I] at :0 - "_waitingReadParamIndexMap: (172, 335)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 2"
[I] at :0 - "waitingParamCount: 2"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "UAVCAN_ENABLE" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "UAVCAN_ENABLE" count: 336 index: 335 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: (172)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 1"
[I] at :0 - "waitingParamCount: 1"
[I] at :0 - "Adding new fact"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[... 40 lines cut ...]
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 172 )"
[I] at :0 - "Closing Port # 200 "COM7""
[I] at :0 - "_linkStatusChanged "113090e0" false"
[I] at :0 - "_linkDisconnected: "\\.\COM7""
[I] at :0 - "link count: 1"
[I] at :0 - "Already disconnected"
[I] at :0 - "Deleting UAS object:  "MAV 001""
[I] at :0 - "After app delete"

ChauvinLou, 2015-06-24:
nsh> param show
Symbols: x = used, + = saved, * = unsaved
    MKBLCTRL_TEST [-1,0] : 0
    PWM_AUX_REV6 [-1,1] : 0
    PWM_AUX_REV5 [-1,2] : 0
    PWM_AUX_REV4 [-1,3] : 0
    PWM_AUX_REV3 [-1,4] : 0
    PWM_AUX_REV2 [-1,5] : 0
    PWM_AUX_REV1 [-1,6] : 0
x   PWM_MAIN_REV8 [0,7] : 0
x   PWM_MAIN_REV7 [1,8] : 0
x   PWM_MAIN_REV6 [2,9] : 0
x   PWM_MAIN_REV5 [3,10] : 0
x   PWM_MAIN_REV4 [4,11] : 0
x   PWM_MAIN_REV3 [5,12] : 0
x   PWM_MAIN_REV2 [6,13] : 0
x   PWM_MAIN_REV1 [7,14] : 0
    FPE_DEBUG [-1,15] : 0
    FPE_SONAR_LP_L [-1,16] : 0.2000
    FPE_SONAR_LP_U [-1,17] : 0.5000
    FPE_LO_THRUST [-1,18] : 0.4000
    RV_YAW_P [-1,19] : 0.1000
    LAUN_THR_PRE [-1,20] : 0.0000
    LAUN_CAT_PMAX [-1,21] : 30.00000
    LAUN_CAT_MDEL [-1,22] : 0.0000
    LAUN_CAT_T [-1,23] : 0.0500
    LAUN_CAT_A [-1,24] : 30.00000
    LAUN_ALL_ON [-1,25] : 0
    ATT_J_EN [-1,26] : 0
    ATT_J33 [-1,27] : 0.0037
    ATT_J22 [-1,28] : 0.0018
    ATT_J11 [-1,29] : 0.0018
x + EKF_ATT_ENABLED [8,30] : 0
    EKF_ATT_V4_R2 [-1,31] : 100.000000
    EKF_ATT_V4_R1 [-1,32] : 10000.00000000
    EKF_ATT_V4_R0 [-1,33] : 0.0008
    EKF_ATT_V3_Q3 [-1,34] : 0.0050
    EKF_ATT_V3_Q2 [-1,35] : 0.0090
    EKF_ATT_V3_Q1 [-1,36] : 0.0800
    EKF_ATT_V3_Q0 [-1,37] : 0.0001
x   ATT_BIAS_MAX [9,38] : 0.0500
x   ATT_ACC_COMP [10,39] : 2
x   ATT_MAG_DECL_A [11,40] : 1
x   ATT_MAG_DECL [12,41] : 0.0000
x   ATT_W_GYRO_BIAS [13,42] : 0.1000
x   ATT_W_MAG [14,43] : 0.1000
x   ATT_W_ACC [15,44] : 0.2000
    BD_OBJ_SURFACE [-1,45] : 0.0031
    BD_OBJ_MASS [-1,46] : 0.6000
    BD_OBJ_CD [-1,47] : 0.1000
    BD_PRECISION [-1,48] : 30.00000
    BD_TURNRADIUS [-1,49] : 120.00000
    BD_GPROPERTIES [-1,50] : 0.0300
x   COM_RC_IN_MODE [16,51] : 0
x   COM_AUTOS_PAR [17,52] : 1
x   COM_RC_LOSS_T [18,53] : 0.5000
x   COM_EF_TIME [19,54] : 10.00000
x   COM_EF_C2T [20,55] : 5.0000
x   COM_EF_THROT [21,56] : 0.5000
x   COM_DL_REG_T [22,57] : 0
x   COM_DL_LOSS_T [23,58] : 10
x   COM_DL_LOSS_EN [24,59] : 0
x   BAT_CAPACITY [25,60] : -1.0000
x + BAT_N_CELLS [26,61] : 6
x   BAT_V_LOAD_DROP [27,62] : 0.0700
x   BAT_V_CHARGED [28,63] : 4.2000
x + BAT_V_EMPTY [29,64] : 3.2000
x   TRIM_YAW [30,65] : 0.0000
x   TRIM_PITCH [31,66] : 0.0000
x   TRIM_ROLL [32,67] : 0.0000
[... 460 lines cut ...]
x   CBRK_IO_SAFETY [334,528] : 0
x   CBRK_RATE_CTRL [335,529] : 0
x + CBRK_SUPPLY_CHK [336,530] : 894281
    CBRK_GPSFAIL [-1,531] : 240024
    CBRK_ENGINEFAIL [-1,532] : 284953
    CBRK_FLIGHTTERM [-1,533] : 121212
    CBRK_AIRSPD_CHK [-1,534] : 0
    CBRK_IO_SAFETY [-1,535] : 0
    CBRK_RATE_CTRL [-1,536] : 0
    CBRK_SUPPLY_CHK [-1,537] : 0
    UAVCAN_BITRATE [-1,538] : 1000000
    UAVCAN_NODE_ID [-1,539] : 1
x   UAVCAN_ENABLE [337,540] : 0
    VT_ARSP_LP_GAIN [-1,541] : 0.3000
    VT_PROP_EFF [-1,542] : 0.0000
    VT_POWER_MAX [-1,543] : 120.00000
    VT_FW_PITCH_TRIM [-1,544] : 0.0000
    VT_FW_PERM_STAB [-1,545] : 0
    VT_MC_ARSPD_TRIM [-1,546] : 10.00000
    VT_MC_ARSPD_MAX [-1,547] : 30.00000
    VT_MC_ARSPD_MIN [-1,548] : 10.00000
    VT_IDLE_PWM_MC [-1,549] : 900
    VT_MOT_COUNT [-1,550] : 0

 551 parameters total, 338 used.

ChauvinLou, 2015-06-24:
Update
Parameter loading using px4 fm4-v2_default.px4 - passed 30 of 30 attempts - failure rate 0%
Parameter loading using firmware-15-06-23-c4d54.px4 passed 11 of 29 attempts - failure rate 42%

ChauvinLou, 2015-06-24:
When using firmware-15-06-23-c4d54.px4

[I] at :0 - "_waitingReadParamIndexMap: (138, 292, 293)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 3"
[I] at :0 - "waitingParamCount: 3"
[I] at :0 - "Adding new fact"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 138 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 292 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 293 )", 

...

DonLakeFlyer, 2015-06-24:
> When using firmware-15-06-23-c4d54.px4

Your firmware has a bug in it which has already been fixed. It needs to be updated to the latest master.

jgoppert, 2015-06-24:
@DonLakeFlyer this was last synced with master yesterday at @PX4/Firmware/a10d34d4a34a82a19e47d74b911ff22402e73409

What change fixed it?

@ChauvinLou it looks like you typed
```
nsh> param
```

Can you try typing:
```
nsh> param index_used 128
```

jgoppert, 2015-06-24:
@ChauvinLou never mind, I see the param show after that, should answer any questions.

DonLakeFlyer, 2015-06-24:
> What change fixed it?

You’ll have to ask Lorenz that. He made the change a while back.

jgoppert, 2015-06-24:
Here is the current comparison of the firmware with master: https://github.com/PX4/Firmware/compare/master...jgoppert:cart_pos_estimator

My guess is that the changes to the cart_pos_estimator branch are exposing an odd bug in the param handling.

jgoppert, 2015-06-24:
@ChauvinLou can you do a param show with the 15-06-23-c4d54.px4 firmware.

ChauvinLou, 2015-06-24:
@jgoppert  using -c4d54 firmware

nsh> param show
Symbols: x = used, + = saved, * = unsaved
    MKBLCTRL_TEST [-1,0] : 0
    PWM_AUX_REV6 [-1,1] : 0
    PWM_AUX_REV5 [-1,2] : 0
    PWM_AUX_REV4 [-1,3] : 0
    PWM_AUX_REV3 [-1,4] : 0
    PWM_AUX_REV2 [-1,5] : 0
    PWM_AUX_REV1 [-1,6] : 0
x   PWM_MAIN_REV8 [0,7] : 0
x   PWM_MAIN_REV7 [1,8] : 0
x   PWM_MAIN_REV6 [2,9] : 0
x   PWM_MAIN_REV5 [3,10] : 0
x   PWM_MAIN_REV4 [4,11] : 0
x   PWM_MAIN_REV3 [5,12] : 0
x   PWM_MAIN_REV2 [6,13] : 0
x   PWM_MAIN_REV1 [7,14] : 0
    FPE_DEBUG [-1,15] : 0
    FPE_SONAR_LP_L [-1,16] : 0.2000
    FPE_SONAR_LP_U [-1,17] : 0.5000
    FPE_LO_THRUST [-1,18] : 0.4000
    RV_YAW_P [-1,19] : 0.1000
    LAUN_THR_PRE [-1,20] : 0.0000
    LAUN_CAT_PMAX [-1,21] : 30.00000
    LAUN_CAT_MDEL [-1,22] : 0.0000
    LAUN_CAT_T [-1,23] : 0.0500
    LAUN_CAT_A [-1,24] : 30.00000
    LAUN_ALL_ON [-1,25] : 0
    ATT_J_EN [-1,26] : 0
    ATT_J33 [-1,27] : 0.0037
    ATT_J22 [-1,28] : 0.0018
    ATT_J11 [-1,29] : 0.0018
x + EKF_ATT_ENABLED [8,30] : 0
    EKF_ATT_V4_R2 [-1,31] : 100.000000
    EKF_ATT_V4_R1 [-1,32] : 10000.00000000
    EKF_ATT_V4_R0 [-1,33] : 0.0008
    EKF_ATT_V3_Q3 [-1,34] : 0.0050
    EKF_ATT_V3_Q2 [-1,35] : 0.0090
    EKF_ATT_V3_Q1 [-1,36] : 0.0800
    EKF_ATT_V3_Q0 [-1,37] : 0.0001
x   ATT_BIAS_MAX [9,38] : 0.0500
x   ATT_ACC_COMP [10,39] : 2
x   ATT_MAG_DECL_A [11,40] : 1
x   ATT_MAG_DECL [12,41] : 0.0000
x   ATT_W_GYRO_BIAS [13,42] : 0.1000
x   ATT_W_MAG [14,43] : 0.1000
x   ATT_W_ACC [15,44] : 0.2000
    BD_OBJ_SURFACE [-1,45] : 0.0031
    BD_OBJ_MASS [-1,46] : 0.6000
    BD_OBJ_CD [-1,47] : 0.1000
    BD_PRECISION [-1,48] : 30.00000
    BD_TURNRADIUS [-1,49] : 120.00000
    BD_GPROPERTIES [-1,50] : 0.0300
x   COM_RC_IN_MODE [16,51] : 0
x   COM_AUTOS_PAR [17,52] : 1
x   COM_RC_LOSS_T [18,53] : 0.5000
x   COM_EF_TIME [19,54] : 10.00000
x   COM_EF_C2T [20,55] : 5.0000
x   COM_EF_THROT [21,56] : 0.5000
x   COM_DL_REG_T [22,57] : 0
x   COM_DL_LOSS_T [23,58] : 10
x   COM_DL_LOSS_EN [24,59] : 0
x   BAT_CAPACITY [25,60] : -1.0000
x + BAT_N_CELLS [26,61] : 6
x   BAT_V_LOAD_DROP [27,62] : 0.0700
x   BAT_V_CHARGED [28,63] : 4.2000
x + BAT_V_EMPTY [29,64] : 3.2000
x   TRIM_YAW [30,65] : 0.0000
[... 455 lines cut ...]
x   SYS_AUTOCONFIG [326,521] : 0
x + SYS_AUTOSTART [327,522] : 12001
x   CBRK_GPSFAIL [328,523] : 240024
x   CBRK_ENGINEFAIL [329,524] : 284953
x   CBRK_FLIGHTTERM [330,525] : 121212
x   CBRK_AIRSPD_CHK [331,526] : 0
x   CBRK_IO_SAFETY [332,527] : 0
x   CBRK_RATE_CTRL [333,528] : 0
x + CBRK_SUPPLY_CHK [334,529] : 894281
    UAVCAN_BITRATE [-1,530] : 1000000
    UAVCAN_NODE_ID [-1,531] : 1
x   UAVCAN_ENABLE [335,532] : 0
    VT_ARSP_LP_GAIN [-1,533] : 0.3000
    VT_PROP_EFF [-1,534] : 0.0000
    VT_POWER_MAX [-1,535] : 120.00000
    VT_FW_PITCH_TRIM [-1,536] : 0.0000
    VT_FW_PERM_STAB [-1,537] : 0
    VT_MC_ARSPD_TRIM [-1,538] : 10.00000
    VT_MC_ARSPD_MAX [-1,539] : 30.00000
    VT_MC_ARSPD_MIN [-1,540] : 10.00000
    VT_IDLE_PWM_MC [-1,541] : 900
    VT_MOT_COUNT [-1,542] : 0

 543 parameters total, 336 used.
nsh>

LorenzMeier, 2015-06-24:
@ChauvinLou As I wrote earlier I need the output of
```
nsh> param index_used 128
```
From the affected Firmware version which doesn't work via MAVLink. I can only help you further with the output of that.

ChauvinLou, 2015-06-24:
@LorenzMeier Using Tera Term  on Windows. Direct connect to Pixhawk using USB cable

nsh> param index_used 128
param: expected a command, try 'load', 'import', 'show', 'set', 'compare', 'select' or 'save'
nsh>

So is there something missing from the "affected Firmware version" build that prevents the "index_used" option from being recognized?

LorenzMeier, 2015-06-24:
Does the index_used command work with any other index? Like 1, 2 or so? If it doesn't work with any index your Firmware is outdated and needs to be updated with the fixes from master.

ChauvinLou, 2015-06-24:
No - as I indicated above it doesn't work with 128 or any other number on either the standard build (px4 fmu-v2_default.px4 nor the c4d54 build.

See note above (6hrs ago) from jgoppert regarding sync time of c4d54 build.

jgoppert, 2015-06-25:
@ChauvinLou  when you say standard build is that the daily build or stable?

ChauvinLou, 2015-06-25:
@LorenzMeier  
Reloaded latest master build - see image

nsh> param index used 128
index 0:     MKBLCTRL_TEST [-1,0] : 0

[image]

ChauvinLou, 2015-06-25:
@LorenzMeier Note the syntax (with underscore) - this is all with c4d54 firmware (output from master is below)
nsh> param index_used 128
param: expected a command, try 'load', 'import', 'show', 'set', 'compare', 'select' or 'save'
nsh>

---------------------------
without underscore
nsh> param index used 128
index 0:     MKBLCTRL_TEST [-1,0] : 0
nsh>

nsh> param index 1
index 1:     PWM_AUX_REV6 [-1,1] : 0
nsh> param index 128
index 128:     FW_LND_USETER [-1,128] : 0

--------------------
With master

nsh> param index_used 1
param: expected a command, try 'load', 'import', 'show', 'set', 'compare', 'select' or 'save'
nsh> param index_used 128
param: expected a command, try 'load', 'import', 'show', 'set', 'compare', 'select' or 'save'
nsh> param index used 1
index 0:     MKBLCTRL_TEST [-1,0] : 0
nsh> param index used 128
index 0:     MKBLCTRL_TEST [-1,0] : 0
nsh> param index 1
index 1:     PWM_AUX_REV6 [-1,1] : 0
nsh> param index 128
index 128:     FW_LND_USETER [-1,128] : 0
nsh>

ChauvinLou, 2015-06-25:
QGroundControl failed parameter load using master firmware on 6th attempt.

Bottom of output:
[I] at :0 - "Received PARAM_VALUE "CBRK_IO_SAFETY" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "CBRK_IO_SAFETY" count: 339 index: 335 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: (330, 336, 337, 338)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 4"
[I] at :0 - "waitingParamCount: 4"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "CBRK_RATE_CTRL" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "CBRK_RATE_CTRL" count: 339 index: 336 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: (330, 337, 338)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 3"
[I] at :0 - "waitingParamCount: 3"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "CBRK_SUPPLY_CHK" QVariant(int, 894281) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "CBRK_SUPPLY_CHK" count: 339 index: 337 mavType: 6 value: QVariant(int, 894281) )"
[I] at :0 - "_waitingReadParamIndexMap: (330, 338)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 2"
[I] at :0 - "waitingParamCount: 2"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "UAVCAN_ENABLE" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "UAVCAN_ENABLE" count: 339 index: 338 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: (330)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 1"
[I] at :0 - "waitingParamCount: 1"
[I] at :0 - "Adding new fact"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Read re-request for (componentId: 50 paramIndex: 330 )"
[I] at :0 - "Closing Port # 200 "COM7""
[I] at :0 - "_linkStatusChanged "112b7400" false"
[I] at :0 - "_linkDisconnected: "\\.\COM7""
[I] at :0 - "link count: 1"
[I] at :0 - "Already disconnected"
[I] at :0 - "Deleting UAS object:  "MAV 001""
[I] at :0 - "After app delete"

---
Tera Term can't connect at this point

-- Remove Lipo
-- Connect using USB only
nsh> param index 330
index 330: x   RC_MAP_PARAM1 [158,330] : 0
nsh>

dogmaphobic, 2015-06-25:
Odd, current master works fine here:
```
nsh> param index_used 128
index 128: x   INAV_SONAR_ERR [128,296] : 0.5000
nsh>
nsh> param index_used
param: no index provided
nsh>
```

It's in param.c:
```
		if (!strcmp(argv[1], "index_used")) {
			if (argc >= 3) {
				do_show_index(argv[2], true);
			} else {
				warnx("no index provided");
				return 1;
			}
		}
```

LorenzMeier, 2015-06-25:
@ChauvinLou I can reproduce this now locally, and I'm looking into it. This is not present on stable, just on master.

LorenzMeier, 2015-06-25:
Ok, fixed on master now:
```
nsh> param index_used 128
index 128: x   INAV_SONAR_ERR [128,296] : 0.5000
```

It was a stupidity introduced on a "cleanup" on master which hadn't been applied to stable.

closed 2015-06-25 by LorenzMeier

ChauvinLou, 2015-06-25:
Test Results after using latest master build.
Three test sequences (power cycle UAV between tests and restart QGR) - all with --full_logging
First test sequence - 15 attempts to connect and load parameter file - 0 failures
Second test sequence - three attempts to connect and load parameter file - third failed to complete - tail of log below
Third test sequence - 15 attempts to connect and load parameter file - fourth attempt failed, 10th passed but QBR reported "operation timeout - no active tranfer" four time.

----
tail of QGR log of test 2 where third attempt failed to complete.
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "CBRK_GPSFAIL" QVariant(int, 240024) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "CBRK_GPSFAIL" count: 339 index: 331 mavType: 6 value: QVariant(int, 240024) )"
[I] at :0 - "_waitingReadParamIndexMap: (332, 333, 334, 335, 336, 337, 338)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 7"
[I] at :0 - "waitingParamCount: 7"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "CBRK_ENGINEFAIL" QVariant(int, 284953) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "CBRK_ENGINEFAIL" count: 339 index: 332 mavType: 6 value: QVariant(int, 284953) )"
[I] at :0 - "_waitingReadParamIndexMap: (333, 334, 335, 336, 337, 338)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 6"
[I] at :0 - "waitingParamCount: 6"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "CBRK_FLIGHTTERM" QVariant(int, 121212) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "CBRK_FLIGHTTERM" count: 339 index: 333 mavType: 6 value: QVariant(int, 121212) )"
[I] at :0 - "_waitingReadParamIndexMap: (334, 335, 336, 337, 338)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 5"
[I] at :0 - "waitingParamCount: 5"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "CBRK_AIRSPD_CHK" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "CBRK_AIRSPD_CHK" count: 339 index: 334 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: (335, 336, 337, 338)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 4"
[I] at :0 - "waitingParamCount: 4"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "CBRK_IO_SAFETY" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "CBRK_IO_SAFETY" count: 339 index: 335 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: (336, 337, 338)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 3"
[I] at :0 - "waitingParamCount: 3"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "CBRK_RATE_CTRL" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "CBRK_RATE_CTRL" count: 339 index: 336 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: (337, 338)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 2"
[I] at :0 - "waitingParamCount: 2"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "CBRK_SUPPLY_CHK" QVariant(int, 894281) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "CBRK_SUPPLY_CHK" count: 339 index: 337 mavType: 6 value: QVariant(int, 894281) )"
[I] at :0 - "_waitingReadParamIndexMap: (338)"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[I] at :0 - "waitingReadParamIndexCount: 1"
[I] at :0 - "waitingParamCount: 1"
[I] at :0 - "Adding new fact"
[I] at :0 - "Received PARAM_VALUE "UAVCAN_ENABLE" QVariant(int, 0) 6"
[I] at :0 - "_parameterUpdate (usaId: 1 componentId: 50 name: "UAVCAN_ENABLE" count: 339 index: 338 mavType: 6 value: QVariant(int, 0) )"
[I] at :0 - "_waitingReadParamIndexMap: ()"
[I] at :0 - "_waitingReadParamNameMap ()"
[I] at :0 - "_waitingWriteParamNameMap ()"
[... 15 lines cut ...]
[I] at :0 - "Manufacturer:  "FTDI""
[I] at :0 - "PortName    :  "COM3" Description :  "Intel(R) Active Management Technology - SOL""
[I] at :0 - "Manufacturer:  "Intel""
[I] at :0 - "PortName    :  "COM10" Description :  "M6e""
[I] at :0 - "Manufacturer:  "CDC Driver""
[I] at :0 - "PortName    :  "COM5" Description :  """
[I] at :0 - "Manufacturer:  """
[I] at :0 - "Configuring port"
[I] at :0 - "CONNECTING LINK:  src\comm\SerialLink.cc 387 type: "" with settings "\\.\COM7" 57600 8 0 1"
[I] at :0 - "_linkStatusChanged "ab3ac28" true"
[I] at :0 - "Temp log "C:/Users/lchauvin/AppData/Local/Temp/FlightData152761.mavlink954312""
[I] at :0 - "TYPE CHANGED TO: 14"
[I] at :0 - "addLink: "ab3ac28""
[I] at :0 - "Add new UAS:  1"
[I] at :0 - "Request to refresh all parameters"
[I] at :0 - "src\uas\UASWaypointManager.cc 1083 "Requesting waypoint list...""
[I] at :0 - "Unknown message from system: 1 message: 87"
[I] at :0 - "src\uas\UASWaypointManager.cc 1083 "Requesting waypoint list...""
[I] at :0 - "Closing Port # 200 "COM7""
[I] at :0 - "_linkStatusChanged "ab3ac28" false"
[I] at :0 - "_linkDisconnected: "\\.\COM7""
[I] at :0 - "link count: 1"
[I] at :0 - "Already disconnected"
[I] at :0 - "Deleting UAS object:  "MAV 001""
[I] at :0 - "After app delete"

DonLakeFlyer, 2015-06-28:
Ok to close now?

ChauvinLou, 2015-07-01:
Much better than before fix, BUT still will not complete transfer 1 out of every 15 or so attempts.
I didn't see this at all on stable (no failures in 30 attempts.